YosysHQ/yosys #331: relative include paths are only relative to root file

robstoddard, 2017-03-13:
I have a project where the digital designer has done some rather wicked things with the includes.  I've found that Yosys does not behave properly when dealing with relative include paths.   I'll give you an example.

The root file is  /path/to/root/file/topfile.sv   and it includes /path/to/fileA/fileA.sv, but uses the relative path ../../fileA/fileA.sv.   This works fine, since the path is with respect to the root file location.

fileA.sv includes /path/to/fileB/fileB.sv... but uses a relative include path ../fileB/fileB.sv instead of the full absolute path.  This does not work fine because Yosys tries /path/to/root/fileB/fileB.sv ...  because the filename that's passed to the preprocessor is the original, root filename.

I've tried to fix this issue but ran into downstream issues.  I'm not sure whether these downstream issues are because Yosys doesn't like the SystemVerilog I'm trying to feed it or there's some problem I've caused in trying to fix this issue...

In my fix, I added `file_push and `file_pop to the preprocessor and a stack of filenames (yes, I know there's one in the lexer too... but that doesn't come into play yet...)  to handle the changes in filename/location.  I also had to change that which gets sent to the `file_push tag, since that's only the most recent relative path...  I needed to either make that absolute or relative to the pwd.  One issue that popped up when I did this was I started getting compiler complaints about not understanding `file_push statements:

ERROR: Parser error in line design/Orchestrate/design/chip.sv:1: Unimplemented compiler directive or undefined macro `file_push.

Which is confusing to me, since that is theoretically handled in the lexer and generated by the preprocessor.   Anyway, if I absorb this in the preprocessor, other issues pop up which have to do with lines of code that now I can't tell what's what... since the compiler lacks filename context.

diff --git a/frontends/verilog/preproc.cc b/frontends/verilog/preproc.cc
index df1005d..2041997 100644
--- a/frontends/verilog/preproc.cc
+++ b/frontends/verilog/preproc.cc
@@ -43,6 +43,8 @@ using namespace VERILOG_FRONTEND;

 static std::list<std::string> output_code;
 static std::list<std::string> input_buffer;
+static std::list<std::string> fn_stack;
+
 static size_t input_buffer_charp;

 static void return_char(char ch)
@@ -291,7 +293,9 @@ std::string frontend_verilog_preproc(std::istream &f, std::string filename, cons

                if (tok == "`include") {
                        skip_spaces();
+                       std::string ofn = fn_stack.back();
                        std::string fn = next_token(true);
+                       std::string ffn;
                        while (fn.size() > 1 && fn[0] == '`' && defines_map.count(fn.substr(1)) > 0)
                                fn = defines_map.at(fn.substr(1));
                        while (1) {
@@ -309,22 +313,43 @@ std::string frontend_verilog_preproc(std::istream &f, std::string filename, cons
                        if (ff.fail() && fn.size() > 0 && fn[0] != '/' && filename.find('/') != std::string::npos) {
                                // if the include file was not found, it is not given with an absolute path, and the
                                // currently read file is given with a path, then try again relative to its directory
+                               ffn = ofn.substr(1, ofn.rfind('/')) + fn;
                                ff.clear();
-                               ff.open(filename.substr(0, filename.rfind('/')+1) + fn);
+                               ff.open(ffn);
+
                        }
                        if (ff.fail() && fn.size() > 0 && fn[0] != '/') {
                                // if the include file was not found and it is not given with an absolute path, then
                                // search it in the include path
-                               for (auto incdir : include_dirs) {
+                               for (auto incdir : include_dirs)
+                               {
                                        ff.clear();
                                        ff.open(incdir + '/' + fn);
-                                       if (!ff.fail()) break;
+                                       ffn = incdir + '/' + fn;
+                                       if (!ff.fail())
+                                       {
+                                               break;
+                                       }
                                }
                        }
                        if (ff.fail())
                                output_code.push_back("`file_notfound " + fn);
                        else
-                               input_file(ff, fn);
+                               input_file(ff, ffn);
+                       continue;
+               }
+
+               if (tok == "`file_push") {
+                       skip_spaces();
+                       std::string name = next_token(true);
+                       fn_stack.push_back(name);
+                       continue;
+               }
+               if (tok == "`file_pop") {
+                       skip_spaces();
+                       fn_stack.pop_back();
                        continue;
                }

robstoddard, 2017-03-13:
Oh, by the way, the code I'm trying this with has been tested in both ModelSim and Verilator.

cliffordwolf, 2017-03-14:
Simple [mcve](https://stackoverflow.com/help/mcve) for this: [include_mcve.zip](https://github.com/cliffordwolf/yosys/files/842004/include_mcve.zip)

cliffordwolf, 2017-03-14:
Fixed in commit 088f9c9.

However, afaics IEEE Std. 1364 does not specify which directory should be the base for relative paths in \`include statements. So I would not count on this to work in all tools. For example, Icarus Verilog fails on the mcve that I posted in my previous message.

closed 2017-03-14 by cliffordwolf