YosysHQ/yosys #332: Yosys install on Linux CentOS

ccalligaro, 2017-03-15:
I'm installing Yosys on Linux CentOS ver 6.8 Final
I have install gcc c++1 and now the compiler release is:
=============================================================
gcc (GCC) 4.8.2 20140120 (Red Hat 4.8.2-15)
Copyright (C) 2013 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
==============================================================
Concerning standard tools the situation is the following:
==============================================================
clang-3.4.2-4.el6.x86_64
bison-2.4.1-5.el6.x86_64
flex-2.5.35-9.el6.x86_64
readline-devel-6.0-4.el6.x86_64
gawk-3.1.7-10.el6_7.3.x86_64
tcl-devel-8.5.7-6.el6.x86_64
libffi-devel-3.0.5-3.2.el6.x86_64
git-1.7.1-4.el6_7.1.x86_64
mercurial-1.4-4.el6.x86_64
graphviz-2.26.0-10.el6.x86_64
xdotool-2.20110530.1-7.el6.x86_64
pkgconfig-0.23-9.1.el6.x86_64
python34-3.4.5-1.el6.x86_64
===============================================================
The last lines of the build output are the following:
===============================================================
gcc -o frontends/verific/verific.o -c   -Wall -Wextra -ggdb -I. -I"./" -MD -D_YOSYS_ -fPIC -I/usr/local/include -std=c++11 -Os -DYOSYS_ENABLE_READLINE -I/usr/lib64/libffi-3.0.5/include   -DYOSYS_ENABLE_PLUGINS -I/usr/include/tcl8.5 -DYOSYS_ENABLE_TCL -DYOSYS_ENABLE_ABC -DYOSYS_ENABLE_COVER frontends/verific/verific.cc
mkdir -p frontends/verilog/
bison -d -r all -b frontends/verilog/verilog_parser frontends/verilog/verilog_parser.y
frontends/verilog/verilog_parser.y:142.21-27: syntax error, unexpected identifier
make: *** [frontends/verilog/verilog_parser.tab.cc] Error 1
===============================================================
It seems to be a problem with bison!

Best regards

Cristiano

OrDavidi, 2017-03-15:
%define parse.error verbose
%define parse.lac full

This syntax was added to bison on bison 2.6+ 
so either install later bison / just delete those lines locally

http://www.gnu.org/software/bison/manual/html_node/_0025define-Summary.html
you can see it here.

Thanks

cliffordwolf, 2017-03-15:
What @OrDavidi said. Those two statements only improve the quality of parser error messaged. You can safely remove them.

closed 2017-03-15 by cliffordwolf

vlsimonks, 2018-11-29:
Hi Daveshah, 

 Is the issue got resolved . Is yes can you please share the procedure to install the yosys in centos.

Thanks,
Vijay.

cliffordwolf, 2018-11-29:
As @OrDavidi said above, just remove those two lines. @vlsimonks