YosysHQ/yosys #333: Problem with $readmemh when passing a file as a parameter

Obijuan, 2017-03-19:
I have define a rom that will be loaded from a generic file:

```verilog
module rom #(
parameter ROMFILE="nofile"
) (
input clk,
input [3:0] A,
output [3:0] D
);

//-- ROM memory
reg [3:0] rom [0:15];

//-- Address bus (4 bits)
wire [3:0] A;

//-- Data bus (8 bits)
reg [3:0] D;

always @(negedge clk) begin
  D <= rom[A];
end

//-- Memory contents read
//-- from the ROMFILE file
initial begin
    $readmemh(ROMFILE, rom);
end
endmodule
```

The top entity instantiates the rom and gives a name to the ROMFILE parameter:

```verilog
`default_nettype none

module main (
 input clk,
 output wire d0, d1, d2, d3
);

 parameter ROMFILE = "rom0.list";

 wire [3:0] addr = 4'h0;
 wire [3:0] data;

 rom #(
  .ROMFILE(ROMFILE)
 ) rom1 (
  .A(addr),
  .D(data),
  .clk(clk)
 );

 assign {d3,d2,d1,d0} = data;

 endmodule
```

When synthesizing with yosys, the following error appears:

```
yosys -p "synth_ice40 -blif hardware.blif" -q rom.v testrom.v
ERROR: Can not open file `nofile` for \$readmemh at rom.v:27.
```
It seems that yosys is trying to open the nofile file, instead of rom0.list

If I create an empty nofile file, it does not shown the error message, and the rom is loaded with the correct rom0.list

It is a minor bug. But we need this feature for using generic roms in icestudio, so that the memory contents can be specify by the user file

cliffordwolf, 2017-03-20:
> It seems that yosys is trying to open the nofile file, instead of rom0.list

It helps when you look at the complete logfile for context. :)

Yosys will first evaluate all modules stand-alone and later, when the hierarchy pass is executed, re-evaluate the modules with the parameter sets they are instantiated with from other modules. This error message is for `rom`, not `main.rom1`. And for `rom` your design says to open the file `nofile`.

Solutions:

1. Read `rom.v` with `read_verilog -defer`. Then Yosys will never try to construct a `rom` top-level entity.

2. Fix your `rom.v`. E.g.:

    parameter ROMFILE=""

    ....

    initial if (ROMFILE) $readmemh(ROMFILE, rom);

closed 2017-03-20 by cliffordwolf

Obijuan, 2017-03-20:
Thanks Clifford!  :-)  It worked!  It was soooo easy to fix (I feel sooo  stupid...)

Thanks a lot for you kind help