YosysHQ/yosys #334: [Fix] Broken builds on GNU/Linux without `<fpu_control.h>`, i.e., glibc

1138-4EB, 2017-03-27:
I'd like to contribute to [icestorm](http://www.clifford.at/icestorm/) providing installing guidelines for [Alpine Linux](https://alpinelinux.org/). This is the preferred distribution for lightweight [docker](https://www.docker.com/) containers, because the base image requires only a few MB: https://www.brianchristner.io/docker-is-moving-to-alpine-linux/ Therefore, it is a sensible choice to provide a relatively compact precompiled "multiplatform" distribution of icestorm + arachne-pr + yosys + iverilog, as opposed to actually cross-compiling each tool.

I have managed to build all the tools but yosys. Precisely, since Alpinex linux does not use glibc (https://en.wikipedia.org/wiki/Alpine_Linux), the following dependency in `libs/minisat/System.h` cannot be met:

``` C
#if defined(__linux__)
#include <fpu_control.h>
#endif
```
My actual workaround is to remove that line:

``` bash
$ git clone https://github.com/cliffordwolf/yosys.git ~/yosys
$ cd ~/yosys && sed -i "/#include <fpu_control.h>/d" libs/minisat/System.h && make -j$(nproc) && make install \
```

However, I think that it is quite an ugly solution.

----

I checked `ENABLE_*` settings in the [Makefile](https://github.com/cliffordwolf/yosys/blob/master/Makefile), but I couldn't identify any related to this feature.

----

An alternative solution is to check `__GLIBC__` (seen [here](http://lists.busybox.net/pipermail/buildroot/2016-February/152266.html)):

``` C
#if defined(__linux__)
#if defined __GLIBC__ 
#include <fpu_control.h>
#endif
#endif
```
@cliffordwolf, if you agree, I will make a PR with this.

----

Furthermore, I don't know at what extent does the removal of `fpu_control` affect performance. It may be possible to use `fenv.h` instead, as shown [here](https://github.com/recalbox/recalbox-buildroot/blob/master/package/jamvm/0001-Use-fenv.h-when-available-instead-of-fpu_control.h.patch). However, I don't feel confident enough to dive into it. Any clue on how many files should be modified will help.

cliffordwolf, 2017-03-27:
Fixed in 71cbe98.

closed 2017-03-27 by cliffordwolf

1138-4EB, 2017-03-27:
Wow, so fast. Thanks a lot.