YosysHQ/yosys #335: Problem with using ice40 bram

niels-moller, 2017-03-27:
The below module can be synthesized using 
yosys -p synth_ice40 -f verilog ram1.vl

----8<-----
module ram1 (input clk,
             input             sel,
             input             we, 
             input [SIZE-1:0]  adr, 
             input [63:0]      dat_i, 
             output reg [63:0] dat_o);
   parameter  SIZE = 5; // Address size

   reg [63:0] mem [0:(1 << SIZE)-1];
   integer    i;

   initial begin
      for (i = 0; i < (1<<SIZE) - 1; i = i + 1)
        mem[i] <= 0;
   end
   
   always @(posedge clk)
     if (sel & ~we)
       dat_o <= mem[adr];

   always @(posedge clk)
     if (sel & we)
       mem[adr] <= dat_i;
endmodule
----8<-----

Output is 

=== ram1 ===

   Number of wires:                  9
   Number of wire bits:            202
   Number of public wires:           6
   Number of public wire bits:     136
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:                  7
     SB_LUT4                         3
     SB_RAM40_4K                     4

so we get to use a SB_RAM40_4K, as intended.

However, when I try merging the two always blocks, I get the following module, which as far as I
understand these things should be equivalent: 

----8<----
module ram2 (input clk,
             input             sel,
             input             we, 
             input [SIZE-1:0]  adr, 
             input [63:0]      dat_i, 
             output reg [63:0] dat_o);
   parameter  SIZE = 5; // Address size

   reg [63:0] mem [0:(1 << SIZE)-1];
   integer    i;

   initial begin
      for (i = 0; i < (1<<SIZE) - 1; i = i + 1)
        mem[i] <= 0;
   end
   
   always @(posedge clk)
     if (sel) begin
       if (~we)
         dat_o <= mem[adr];
       else
         mem[adr] <= dat_i;
     end
endmodule
----8<----

synthesis using yosys -p synth_ice40 -f verilog ram2.vl gives

=== ram2 ===

   Number of wires:               1684
   Number of wire bits:           3893
   Number of public wires:          38
   Number of public wire bits:    2184
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:               3821
     SB_DFFE                      2112
     SB_LUT4                      1709

So the ram isn't synthesized using SB_RAM40_4K, and synthesis is also a lot slower due to the much larger net.

In the logs, the first difference which looks interesting to me is for the MEMORY_DFF pass, with

Checking cell `$memwr$\mem$ram1.vl:23$104' in module `\ram1': merged $dff to cell.
Checking cell `$memrd$\mem$ram1.vl:19$67' in module `\ram1': merged data $dff with rd enable to cell.

for ram1.vl, and

Checking cell `$memwr$\mem$ram2.vl:22$101' in module `\ram2': merged $dff to cell.
Checking cell `$memrd$\mem$ram2.vl:20$69' in module `\ram2': no (compatible) $dff found.

for ram2.vl. If I read that correctly, it means yosys can't recognize the RAM read port in ram2.vl.

I would expect yosys to treat both versions identically, and generate the proper enable signals (sel & we for write, sel & ~we for read) no matter if it's expressed in one or two always block.

And nevertheless, yosys really is an awesome tool!

Best regards,
/Niels

PS. Apologies if formatting looks crappy, this is the first time ever I file an issue using github, and 
me and web forms usually have some difficulty to get on friendly terms.

niels-moller, 2017-03-27:
Ooops, forgot to mention the version of yosys I'm using. I've pulled and built the latest version from the git repo:

$ yosys -V
Yosys 0.7+161 (git sha1 58ee8e3, gcc 6.3.0-10 -fPIC -Os)

Installed from source on an x86_64 debian gnu/linux system.

Regards,
/Niels

piranna, 2017-04-24:
It seems I'm having the same problem :-/ My code is at https://github.com/piranna/wasmachine/blob/master/src/stack.v

piranna, 2017-04-27:
I've found the next output on the console:

```
7.2.4. Executing AST frontend in derive mode using pre-parsed AST for module `\SuperStack'.
Parameter \WIDTH = 29
Parameter \DEPTH = 3
Generating RTLIL representation for module `$paramod\SuperStack\WIDTH=29\DEPTH=3'.
Warning: Replacing memory \stack with list of registers. See src/SuperStack.v:47, src/SuperStack.v:97
```

[SuperStack](https://github.com/piranna/wasmachine/blob/master/src/SuperStack.v) is an improved stack, that add configurable underflow limit and posibility to set values to specific positions, base code is the same as the already related `stack`. It's interesting that these lines are exactly

```
out <= stack[index-1];
stack[index] = data;
```

piranna, 2017-04-27:
@cliffordwolf thanks for the link, I'll take a look. By the way, according to https://www.reddit.com/r/yosys/comments/2kgfdr/incorrect_synthesis_results_for_multiport/cln0vnm/ (your own post :-) ) seems this error would happen too when accessing to memory in a way that `yosys` memory model is not ready to. As you can see, I'm having too wide slices on my stack, would that be (one of) the reason(s)? That would be curious, because on the test I'm using regular 8 bit slices and it's also not using the BRAMs... :-/

```
7.2.3. Executing AST frontend in derive mode using pre-parsed AST for module `\stack'.
Parameter \WIDTH = 51
Parameter \DEPTH = 1
Generating RTLIL representation for module `$paramod\stack\WIDTH=51\DEPTH=1'.

7.2.4. Executing AST frontend in derive mode using pre-parsed AST for module `\SuperStack'.
Parameter \WIDTH = 29
Parameter \DEPTH = 3
Generating RTLIL representation for module `$paramod\SuperStack\WIDTH=29\DEPTH=3'.
Warning: Replacing memory \stack with list of registers. See src/SuperStack.v:47, src/SuperStack.v:97

7.2.5. Executing AST frontend in derive mode using pre-parsed AST for module `\SuperStack'.
Parameter \WIDTH = 66
Parameter \DEPTH = 7
Generating RTLIL representation for module `$paramod\SuperStack\WIDTH=66\DEPTH=7'.
Warning: Replacing memory \stack with list of registers. See src/SuperStack.v:47, src/SuperStack.v:97
```

piranna, 2017-04-28:
Thanks @cliffordwolf, honestly a really interesting link, I'm very new to Verilog and this kind of "best practices" texts are really useful to me :-) I'm too much used to reuse variables and calculations as I use to do on regular languages, so I have a question: is it better to convert things like

```verilog
index = index - (1+data);

status <= index ? `NONE : `EMPTY;
```

to

```verilog
index <= index - (1+data);

status <= index - (1+data) ? `NONE : `EMPTY;
```

although the `index - (1+data)` fragment is duplicated? And what happens with things like this?

```verilog
index <= index + 1;

stack[index] <= data;
```
I know that on nonblocking assignements right-hand will be calculated when they are reached, but what happens with left-hand? Will the data being written to offset `index` (previous index value), or to offset `index+1` (new index value)?

cliffordwolf, 2017-04-28:
@piranna The data will be written to `stack[index]` (previous index), not `stack[index+1]`.

eddiehung, 2019-06-25:
Looks like #1130 fixes this as well. OP testcase added in ab6e8ce.

closed 2019-06-25 by eddiehung