YosysHQ/yosys #338: Feature suggestion: search path for `readmemb`

thoughtpolice, 2017-04-11:
**Problem**: I am generating some Verilog code, and putting it under a directory called `build/verilog/Foo`. (All build products go under `build`.) There are several files here, and one uses `$readmemb` in order to initialize some BRAMs from a `.hex` file. The `.hex` file is also inside `build/verilog/Foo` next to the Verilog. The verilog file simply instantiates the file using `$readmemb("file.hex",...);`, i.e. it is all relative to `build/verilog/Foo`.

However, I would like to execute `yosys` in the *top level directory*, not under `build/Verilog/Foo`. However, if I do this, I get:

```
yosys> read_verilog build/verilog/Foo/*.v

2. Executing Verilog-2005 frontend.
Parsing Verilog input from `build/verilog/Foo/Foo_blockRamFile.v' to AST representation.
Generating RTLIL representation for module `\Foo_blockRamFile'.
ERROR: Can not open file `./unit.hex` for \$readmemb at build/verilog/Foo/Foo_blockRamFile.v:29.
```

I asked about this on IRC. However, looking [at the code](https://github.com/cliffordwolf/yosys/blob/f3324ed0cc35590e4d0042f6265b83bab0af37bf/frontends/ast/simplify.cc#L2578), this is roughly to be expected: `ifstream` will not resolve this for you, and the parser simply looks at the cell in order to extract the file name and pass it to `ifstream::open()`.

---

I have currently worked around this by instead A) making all paths in my synthesis scripts absolute (based on the current build directory) and B) making my build `cd` into the right place before calling `yosys`.

However, I'm wondering if there's any interest, perhaps, in some kind of "search path" option, so that Yosys can look for such files in another directory. This is perhaps a bit narrow and specific, so a more general concept of "search path" (including, but not limited to, `$readmemb` & co) may be more appropriate.

jiegec, 2021-01-31:
I encountered the same problem. Any way to fix?

thoughtpolice, 2021-02-04:
No, but this is definitely still an issue for me. I'm willing to write a patch for it, though. Maybe an option to `read_verilog` or something would be the most appropriate path forward...

lethalbit, 2022-08-16:
As of writing, commit c26b2bf5 `$readmem` has [logic](https://github.com/YosysHQ/yosys/blob/c26b2bf543a226e65a3fb07040bb278d668accf2/frontends/ast/simplify.cc#L4306) for trying first in the tool CWD followed by checking local to where the file is.

Tentatively closing as resolved as with a test case similar to the original post's example it works as expected.

closed 2022-08-16 by lethalbit