YosysHQ/yosys #343: Equivalence check issue with hierarchy

rafaeltp, 2017-05-01:
Hi,

I am having an issue with equivalence check when my gate module have multiple levels of hierarchy. The sample code is:

```
module gold(input i1, input i2, output o1);
 assign o1 = i1 & i2;
endmodule

module in_w(i1, i2, o1) ;
  input i1, i2;
  output o1;
  assign o1 = (i1 & i2);
endmodule

module top_w( i1, i2, o1);
  input i1, i2;
  output o1;
  in_w dut(.in1(in1), .in2(in2), .o1(o1));
endmodule
```

I am using the following script:

```
#flatten
equiv_make gold top_w equiv
#flatten
hierarchy -top equiv
opt -purge
equiv_simple
equiv_status -assert
```

I tried flatten in both noted places, and not using it at all, and in all the three cases the equiv_status failed.
Comparing with in_w yields a successful comparison.

```
equiv_make gold in_w equiv
hierarchy -top equiv
opt -purge
equiv_simple
equiv_status -assert
```

cliffordwolf, 2017-05-13:
The problem is that `in1` and `in2` in your `top_w` are bogus. It should be `i1` and `i2`.

After reading the design files you should always run something like `prep` or at least `hierarchy -check; proc; clean` to do the necessary high-level processing of your design. This would have yielded the following error for your design:

```
2. Executing HIERARCHY pass (managing design hierarchy).
ERROR: Module `in_w' referenced in module `top_w' in cell `dut' does not have a port named 'in2'.
```

The following script works for me with the fixed design:

```
read_verilog test.v
prep
equiv_make gold top_w equiv
hierarchy -top equiv
flatten
opt -purge
equiv_simple
equiv_status -assert
```

And this is the fixed design (`test.v`):

```
module gold(input i1, input i2, output o1);
 assign o1 = i1 & i2;
endmodule

module in_w(i1, i2, o1) ;
  input i1, i2;
  output o1;
  assign o1 = (i1 & i2);
endmodule

module top_w( i1, i2, o1);
  input i1, i2;
  output o1;
  // in_w dut(.in1(in1), .in2(in2), .o1(o1));
  in_w dut(.i1(i1), .i2(i2), .o1(o1));
endmodule
```

closed 2017-05-13 by cliffordwolf

rafaeltp, 2017-05-13:
Shoot, I can't believe I missed that, thanks for the feedback, I'll keep that in mind for the future.