YosysHQ/yosys #347: How to use write_aiger

5nizza, 2017-05-29:
hi,
How to use `write_aiger` correctly?

When I call it, it crashes with
```
6. Executing AIGER backend.
ERROR: Unsupported cell type: $_ORNOT_ ($abc$646$auto$blifparse.cc:346:parse_blif$648)
```

The full sequence of commands I use is:
```
./yosys
read_verilog ./tests/simple/fsm.v
flatten
write_aiger /tmp/hoho.aag
```

(I use the latest version from the repo). Thanks!

cliffordwolf, 2017-05-29:
> When I call it, it crashes with

It's not crashing. It's quitting with an error message. That's a very very different thing. A program that is crashing is doing something wrong, i.e. the guy who programmed it did something wrong. A program that terminates with an error message is telling the user that they are doing something wrong.

Accusing programmers of having written a buggy program, when in fact the program is telling you that you are using it incorrectly, can be a very effective way of making programmers angry. I recommend avoiding this scenario. :)

> ERROR: Unsupported cell type: $_ORNOT_ ($abc$646$auto$blifparse.cc:346:parse_blif$648)

Quoting the `write_aiger` help message (that is printed by `help write_aiger`):

> [..] The design must be flattened and must not contain any cell types except $_AND_, $_NOT_, simple FF types, $assert and $assume cells, and $initstate cells. [..]

Thus the error message: `$_ORNOT_` is not one of those cell types. Use for example `abc` to map the design to AND and NOT gates only.

> The full sequence of commands I use is:

I seriously doubt that. The script you have posted does not call ABC, neither directly nor indirectly. But the cell name `$abc$646$auto$blifparse.cc:346:parse_blif$648` is a name generated by the ABC pass, and nothing else in Yosys currently maps anything to ORNOT gates. There is no way this script produced the error message you posted. Oh, and since `write_aiger` is the third command in that script, the aiger backend log message would be prefixed with `3.`, not `6.`.

But looking at the script you posted: You do not seem to bother at all to run `hierarchy`, `proc` or anything else that resembles a synthesis script. I'm not sure what exactly you expect the tool to do here. You might want to read some of the Yosys documentation, or at least [this post on stack overflow](https://stackoverflow.com/questions/31434380/what-is-a-good-template-yosys-synthesis-script).

The following script does what I think you want to do:

```
read_verilog ./tests/simple/fsm.v
synth -flatten -top fsm_test
abc -g AND
write_aiger -ascii hoho.aag
```

Notice that I first bring the design all the way from the output of the HDL front-end into a gate-level representation by calling `synth`, then I convert the gate level representation into an explicit and-inverter graph circuit by mapping it to AND and NOT gates using the abc command and only then I write it to the aiger file.

In your script you have a .aag file, so I assume you are interested in generating the ASCII variant of the AIGER file format. You have to call `write_aiger` with the `-ascii` option to generate this format.

closed 2017-05-29 by cliffordwolf

5nizza, 2017-05-30:
wow, you seemed to be pissed off with "crashes".
Whatever, thanks for providing the detailed, precise, and helpful answer!
(all I wanted is to convert verilog into aiger)

cliffordwolf, 2017-05-30:
> wow, you seemed to be pissed off with "crashes".

It's definitely a communication pattern that triggers something.. :)

> Whatever, thanks for providing the detailed, precise, and helpful answer!
(all I wanted is to convert verilog into aiger)

Glad that I could help.

2994186010, 2021-02-18:
When I execute the script file

`read_verilog ./tests/simple/fsm.v
synth -flatten -top fsm_test
abc -g AND
write_aiger -ascii hoho.aag`

I encounter the error as following
`ERROR: Unsupported cell type: $_SDFF_PP0_ ($auto$simplemap.cc:527:simplemap_adff_sdff$289)`

What can I do to solve the question?

gustad, 2021-06-14:
I'm also getting the same error:

abc -dff -g AND
write_aiger -ascii -symbols -zinit out.aag

[00000.270353] ERROR: Unsupported cell type: $_SDFF_PP0_ ($auto$simplemap.cc:527:simplemap_adff_sdff$87)

mwkmwkmwk, 2021-06-14:
Run `dffunmap` before `abc`.

gustad, 2021-06-15:
> Run `dffunmap` before `abc`.

Thanks! Seems like this is a new feature as I did not have this in my scripts which worked with an older yosys version.

christos-vasileiou, 2023-03-16:
> > Run `dffunmap` before `abc`.
> 
> Thanks! Seems like this is a new feature as I did not have this in my scripts which worked with an older yosys version.

Can you please note down the yosys version that you were able to run and map to an AIG?

gustad, 2023-03-19:
> Can you please note down the yosys version that you were able to run and map to an AIG?

One of my yosys generated aiger files have  this comment:
```
c
Generated by Yosys 0.9+932 (git sha1 304e5f9e, clang 6.0.1 -fPIC -Os)

```