YosysHQ/yosys #348: Produced AIGER file contains two different names for the same latch

5nizza, 2017-05-30:
hi again,
Consider the following verilog file:

[verilog.v.zip](https://github.com/cliffordwolf/yosys/files/1038949/verilog.v.zip)

When translating into AIGER by executing 
```
read_verilog verilog.v
synth -flatten
abc -g AND
write_aiger -ascii -symbols hoho.aag
```

the resulting AIGER assigns two different names for latch `l43` (in the symbols table)
[hoho.aag.zip](https://github.com/cliffordwolf/yosys/files/1038959/hoho.aag.zip)
 
```
l43 __qdead
o0 bad
l43 bad
```

I don't know if that is a bug or not, but original AIGER tools from Armin fail on such files, which is strange.
Thanks.

cliffordwolf, 2017-05-30:
The wires `bad` and `__qdead` definitely are the same net, so it seems like it is correct for this latch to have two symbols:

    assign bad = __qdead;

> but original AIGER tools from Armin fail on such files

Interesting. I've now changed the behavior of `write_aiger` in c365e33 to consolidate multiple symbols for the same object and print them in one line (with blanks separating the individual symbols). At least `aigbmc` from the AIGER package seems to accept this syntax.

Looking at your verilog input I would guess that you would like to use this to create specifications for reactive synthesis problems (some of your inputs have names that start with `controllable_`). Cool! Afaict the new behavior should be compatible with the "Extended AIGER Format for Synthesis", since it is possible that a latch or output has multiple symbols, but if an input has multiple symbols that would indicate a driver-driver conflict and thus an invalid circuit specification, therefore it should be guaranteed that all `i*` symbols only have one name. (So the `controllable_*` names will never be accidentally prefixed with another symbol name followed by a blank.)

closed 2017-05-30 by cliffordwolf

5nizza, 2017-05-31:
thanks, it works now!
(yes, i hopefully use if for my tool at SYNTCOMP, but in a very simple scenario --- to generate AIGER for a state machine described in verilog)