YosysHQ/yosys #349: Unexpected OP_NAND

larsbrinkhoff, 2017-06-01:
Is there a problem with this?

    module cpu (input wire x, input wire y, output reg z);
       always @*
         z <= x ~& y;
    endmodule

I get this error message:

    -- Running command `read_verilog verilog/cpu.v; synth_ice40 -blif nybble.blif'
    1. Executing Verilog-2005 frontend.
    Parsing Verilog input from `verilog/cpu.v' to AST representation.
    ERROR: Parser error in line verilog/cpu.v:3: syntax error, unexpected OP_NAND

cliffordwolf, 2017-06-01:
Ooops! Thanks for reporting this. Fixed in 129984e.

PS: That's the smallest cpu module I've ever seen! :)

closed 2017-06-01 by cliffordwolf

larsbrinkhoff, 2017-06-01:
Thanks!

It's kind of an extract from my toy stack CPU.  But then, with just 11 instructions that's quite small too.  I picked NAND as the only bitwise logical instruction, hence this bug report.