YosysHQ/yosys #350: Wrong output when converting Verilog to AIG

kleinreact, 2017-06-16:
I try to convert verilog to AIG's using the following script (if there is any better or more recommended way to do this, please let me know):
```
read_verilog <file.in>
synth
abc -clk clk -g AND
opt
write_aiger -ascii -symbols <file.out>
```
However, for the following test file:
```
module test
  ( input clk
  , input in
  , output out
  );

reg [1:0] state = 0;

always @(posedge clk)
  case (state)
    0:
      if (!in || in)
        state <= 1;
      else 
        state <= state;

    1:
      if (in)
        state <= 1;
      else 
        if (!in)
          state <= 1;
        else 
          state <= state;
  endcase

always @(*)
  case (state)
    0:
      out = 0;

    1:
      begin
         if (in)
           out = 1;

         if (!in)
           out = 0;
      end
  endcase
   
endmodule
```
the following output gets generated:
```
aag 2 2 0 1 0
2
4
4
i0 clk
i1 in
o0 out
c
Generated by Yosys 0.7+196 (git sha1 8f8bacc, gcc 6.3.0 -fPIC -Os)
```
I don't think this is correct.

cliffordwolf, 2017-06-18:
The problem here is that `abc -dff` (and `abc -clk ...`) do not preserve initial values at the moment. If you ignore the initial value assignment for `state` it is safe to assume constant `state=1` with implies `out=in`. Here is an even simpler design which demonstrates this issue:

```
module demo (input clk, input i, output o);
  reg q = 0;
  always @(posedge clk) q <= 1;
  assign o = q & i;
endmodule
```

I will look into preserving initial values in `abc -dff` and `abc -clk ...`. Should not be to hard to do.

cliffordwolf, 2017-06-20:
Fixed in commit 155a80d.

closed 2017-06-20 by cliffordwolf