YosysHQ/yosys #351: No history on error exit

kleinreact, 2017-06-16:
Is it desired that the call history is not saved if yosys exits through an error in interactive mode? I find this very annoying when testing/developing a script, as I then have to restart from scratch, where I need to retype all previous commands. 

In general, I do not understand why the interactive mode is exited on the error at all. Wouldn't it be more useful if just the error message is printed an yosys then returns into its last valid state (just before the erroneous command was executed)?

AlexDaniel, 2017-06-16:
Yes, this is what I never understood also. Why exit on error? There could be a reason for this, but it is pretty damn inconvenient…

cliffordwolf, 2017-06-18:
Internally in Yosys there are two functions that can be used to raise an error log_error() and log_cmd_error(). When the latter is used the caller guarantees that the internal state is still a valid state for a design to be in, and in interactive mode yosys will return to the command prompt in response to this call. However, I'm a lazy guy and when I can avoid thinking about something I usually do so, so in most cases I simply call log_error() instead and don't need to bother to think about the state I'll leave the design in.

In many cases it should be simple to convert code that calls log_error() to call log_cmd_error() instead. If you provide a list of candidates that annoy you the most I'll be happy to look into those cases.

Yes, that the history is not saved on error is annoying. I'll look into that.

kleinreact, 2017-06-19:
For me, the most annoying ones are `abc` and `write_aiger`. 

I have not checked the other `write` commands so far, but I think they should have high priority too, as they usually are used to test and debug a script.

cliffordwolf, 2017-06-20:
Commit f6421c8 converts the `abc` "clock domain not found" message to log_cmd_error(). Afaics that's the only candiate for that in the `abc` command, all the other calls to log_error() are in response to things like filesystem IO errors.

Commits c0ca994 and 1f517d2 add writing of history file in case of an unrecoverable error.

kleinreact, 2017-06-20:
Thanks for fixes so far. They already help a lot.

Regarding the IO errors, why exactly are they excluded of using `log_cmd_error`? Lets say I write something incomplete like this
```
read_verilog <file>
synth 
abc -script write_aiger
```
This triggers an error and I have to start from scratch. I mean, in general, it may always happen that one misses some command or gets something wrong by accident. Still, I wouldn't expect that the program completely crashes in interactive mode.

cliffordwolf, 2017-06-20:
> abc -script write_aiger

I don't think that will be an IO error in Yosys. We simply pass the script file name to ABC, which will then fail. In yosys that is caught as this error:

```
log_error("ABC: execution of command \"%s\" failed: return code %d.\n", buffer.c_str(), ret);
```

At this point we have already ripped all the logic out of the design. If we'd return to the command prompt we would be left with a design that is missing all the logic that was passed to the call to ABC that failed. Returning to a reasonable state at this point would require major rewrites of the ABC pass in Yosys.

closed 2019-03-04 by cliffordwolf