YosysHQ/yosys #357: AIGER writer: outputs are optimized away

5nizza, 2017-07-04:
hi Clifford,
you seemed to have fixed one such case, here is another one:      
verilog file contains two outputs while the produced AIGER has only one.      
Verilog:
```
module model(cancel, req, grant, ack);

input cancel;
input req;

output grant;
output ack;

reg [0:0] __m;

wire grant;
wire ack;

assign grant = ((cancel==0) && (__m==0) && (req==0)) || ((cancel==1) && (__m==0) && (req==0)) || ((cancel==1) && (__m==0) && (req==1)) || ((cancel==0) && (__m==0) && (req==1));
assign ack = ((cancel==0) && (__m==0) && (req==0)) || ((cancel==1) && (__m==0) && (req==0)) || ((cancel==1) && (__m==0) && (req==1)) || ((cancel==0) && (__m==0) && (req==1));

initial begin
__m = 0;
end

always@($global_clock)
begin
if (((cancel==0) && (__m==0) && (req==0))) __m <= 0;
else if (((cancel==1) && (__m==0) && (req==0))) __m <= 0;
else if (((cancel==1) && (__m==0) && (req==1))) __m <= 0;
else if (((cancel==0) && (__m==0) && (req==1))) __m <= 0;
end
endmodule
```

The translation script I use is:
```
read_verilog ./missing_outs.v
synth -flatten -top model
abc -g AND
write_aiger -ascii -symbols ./missing_outs.aag
```
The produced AIGER misses one output in the header and its definition:
```
aag 3 2 1 1 0
2
4
6 6
7
i0 cancel
i1 req
l0 __m
o0 ack grant
c
Generated by Yosys 0.7+214 (git sha1 621787a, clang 3.8.0-2ubuntu4 -fPIC -Os)
```

cliffordwolf, 2017-07-04:
Grant and ack are the same AIG node. If you remember, in response to #348 I added a feature that outputs all names for an AIG node separated by whitespaces in one label because some tools don't like it if two label lines are generated for the same AIG node.

5nizza, 2017-07-04:
yes, I remember that issue: Armin's `aigtoaig` will complain if there is a comment like
```
l0 name
l0 another_name
```
but here we have comments about "different" outputs -- we could have smth like `o0 ack` and `o1 grant` (event though those outputs refer to the same AIG node).      
Here is the "expected" AIGER output:
```
aag 3 2 1 2 0
2
4
6 6
7
7
i0 cancel
i1 req
l0 __m
o0 ack
o1 grant
c
some comment
```
thanks.

cliffordwolf, 2017-07-05:
Implemented in 5442554. That AIGER back-end is not getting any simpler.. :)

closed 2017-07-05 by cliffordwolf