YosysHQ/yosys #358: Coding question: How to pass object to select command.

Kmanfi, 2017-07-05:
Hi,
how to make named selection of the current cell instance?
```
for (auto cell: module->selected_cells())
{
    Pass::call(yosys_design, "select -set current_cell ..??..");
}
```

cliffordwolf, 2017-07-05:
Untested code ahead (might contain subtle bugs):

```
for (auto cell : module->selected_cells()) {
    RTLIL::Selection mysel(false);
    mysel.select(module, cell);
    design->selection_vars["current_cell"] = mysel;
}
```

Not quite sure what you want to do with that. You might be interested to know that there is a Pass::call_on_selection() method:

```
for (auto cell : module->selected_cells()) {
    RTLIL::Selection mysel(false);
    mysel.select(module, cell);
    Pass::call_on_selection(design, mysel, "...");
}
```

closed 2017-07-05 by cliffordwolf

Kmanfi, 2017-07-05:
```
for (auto cell : module->selected_cells()) {
    RTLIL::Selection mysel(false);
    mysel.select(module, cell);
    design->selection_vars["current_cell"] = mysel;
}
```
Above version doesn't work as expected. It will give assert. Any idea how to 
ERROR: Assert `p[0] == '$' || p[0] == '\\'' failed in rtlil.h:104.

Const char *p has value "current_cell"
```
		static inline int get_reference(const char *p)
		{
			log_assert(destruct_guard.ok);

			if (p[0]) {
				log_assert(p[1] != 0);
				log_assert(p[0] == '$' || p[0] == '\\');
			}

```

cliffordwolf, 2017-07-05:
I said there might be subtle bugs! :)

This version should work:

    design->selection_vars["\\current_cell"] = mysel;