YosysHQ/yosys #359: bad error on valid syntax

OrDavidi, 2017-07-05:
This is a valid verilog as far as i know : 

assign b= a==4'd1? 1'b0 : 1'b1;

now, the question mark is not spaced from the decimal number ( this is ok because "?" is not a valid char when dealing with decimal values only on hex / octa / binary - i checked the spec to be sure )

and i get this error : 
const2ast.cc:52  Invalid use of [a-fxz?] in decimal constant at .. 

i guess it should be handle in the lexer 

**current code : verilog_lexer.l:219** 

[0-9]*[ \t]*\'s?[bodhBODH][ \t\r\n]*[0-9a-fA-FzxZX?_]+ {
	frontend_verilog_yylval.string = new std::string(yytext);
	return TOK_CONSTVAL;
}

**my suggestion :** 

current code : verilog_lexer.l:219 
[0-9]*[ \t]*\'s?[bohBODH][ \t\r\n]*[0-9a-fA-FzxZX?_]+ {
	frontend_verilog_yylval.string = new std::string(yytext);
	return TOK_CONSTVAL;
}
current code : verilog_lexer.l:219 
[0-9]*[ \t]*\'s?d[ \t\r\n]*[0-9_]+ {
	frontend_verilog_yylval.string = new std::string(yytext);
	return TOK_CONSTVAL;
}

Thanks

cliffordwolf, 2017-07-05:
I've now tested this with Verific, Verilator, Modelsim, XSim, and Vivado. They all say this is a syntax error:

```
Verific:
VERIFIC-ERROR [VERI-1868] test.v:3: illegal use of x or z character in decimal number
VERIFIC-ERROR [VERI-1137] test.v:3: syntax error near 1
VERIFIC-ERROR [VERI-1072] test.v:4: module test ignored due to previous errors

Verilator:
%Error: test.v:3: Mixing X/Z/? with digits not legal in decimal constant: 1?
%Error: test.v:3: syntax error, unexpected INTEGER NUMBER, expecting ',' or ';'
%Error: Exiting due to 2 error(s)

Modelsim:
** Error: (vlog-13064) test.v(3): near "1": X and Z values not allowed in decimal constants.
** Error: (vlog-13069) test.v(3): near "1": syntax error, unexpected INTEGER NUMBER, expecting ';' or ','.

XSim:
ERROR: [VRFC 10-895] illegal use of x or z character in decimal number [/home/clifford/Work/yosys/test/test.v:3]
ERROR: [VRFC 10-1412] syntax error near 1 [/home/clifford/Work/yosys/test/test.v:3]
ERROR: [VRFC 10-1040] module test ignored due to previous errors [/home/clifford/Work/yosys/test/test.v:1]

Vivado:
ERROR: [Synth 8-2715] syntax error near 1 [/home/clifford/Work/yosys/test/test.v:3]
```

It works with all those tools (and Yosys) when a blank is inserted before the `?`.

closed 2017-07-05 by cliffordwolf

OrDavidi, 2017-07-05:
I think synopsys design compiler is ok with it.
But if this is the case i think it is good enough to act like other tools :)

Thanks!