YosysHQ/yosys #361: Width of alu and macc cells is not reported

heiner-bauer, 2017-07-13:
Using 'stat -width' does not include widths for $alu and $macc cells.
Is there a specific reason why they are not part of the list starting in stat.cc:109 ?

cliffordwolf, 2017-07-14:
Commit  c00d8a5 adds `$alu` cells to the list.

There is no good way to create a single meaningful "width" value for `$macc` cells. The sum of 64 2x2 multiplications, and a single 128x128 multiplication have the same data port widths but vastly different complexity.

closed 2017-07-14 by cliffordwolf

heiner-bauer, 2017-07-14:
Right. I didnt look at the cell implemenation of $macc before. I figure the additional 'c' stands for 'configurable'?
> The sum of 64 2x2 multiplications, and a single 128x128 multiplication have the same data port 

True, but shouldn't the (minimum required) output port width differ? The first should have something in the order of 2x2+8+1 bits while the latter should have 2x128 bits, if I understand the cell functionality correctly (I saw that further documentation for this cell is already on your to-do list).

cliffordwolf, 2017-07-14:
> I figure the additional 'c' stands for 'configurable'?

Actually no. The "acc" just stands for accumulate. :) But it is a very flexible cell type..

> True, but shouldn't the (minimum required) output port width differ?

True for the simple example I gave. But consider this (run through something like `yosys -p 'prep; alumacc;; dump' test.v`):

```
module test (input [63:0] A, B, C, D, input [127:0] E, F, output reg [127:0] X, Y); 
        integer i;
        always @* begin
                X = A*B + E;
                Y = F;
                for (i = 0; i < 64; i=i+1)
                        Y = Y + C[i]*D[i];
        end
endmodule
```

This will create a design with two `$macc` cells that have the same port sizes but vastly different complexity.