YosysHQ/yosys #370: single bit foor loop increments

heiner-bauer, 2017-08-03:
While elaborating the Verilog module below, Yosys rapidly hogs memory (multiple GB) and does not seem to finish. If the counter increment is changed to the examples in the comment, everything works fine. Although it can be easily avoided, I assume `i = i + 1'b1` is still valid syntax and hence should be supported by Yosys.
``` Verilog
module yosys_loop_increment(trigger_i, data_o);

   input trigger_i;
   output [7:0] data_o;

    parameter DEPTH = 5;
   integer i;

   always @(*) begin
      data_o = 8'b0;
      if(trigger_i) begin
         // working: i = i + 1 or i = i + 2'b1 or i = i + 32'b1 or i = i + 'b1
         for(i = 0; i < DEPTH; i = i + 1'b1) begin
            data_o = data_o + i;
         end
      end
   end
          
endmodule
```

cliffordwolf, 2017-08-03:
The problem here is that for some reason 1'b1 is sign extended to a 32 bit -1, and then the whole loop is counting down instead of up. I have no idea why this happens. I'll investigate..

cliffordwolf, 2019-04-22:
Afaict #952 fixes this issue.

PR #952 (2019-04-22): Determine correct signedness and expression width in for-loop unrolling
Fixes #370
commit 4ad0ea5c (2019-04-22): Determine correct signedness and expression width in for loop unrolling, fixes #370
closed 2019-04-22 by cliffordwolf