YosysHQ/yosys #372: Feature request: Pass to optimize $DFF_xy with constant R input to $DFF_x

azonenberg, 2017-08-06:
Example: DFF_PN0 with R=1 is equivalent to a DFF_P.

Not doing this leads to silly results like

    always @(posedge clk_108hz or negedge 1'h1)
    if (!1'h1)
      \shreg.shreg [0] <= 0;
    else
      \shreg.shreg [0] <= led_lfosc_ff;

azonenberg, 2017-08-06:
If this doesn't already exist I'll write one, it seems pretty straightforward. If there's already some way to do this, please let me know how - none of the optimizations I've tried remove the redundant reset.

cliffordwolf, 2017-08-06:
Implemented in c4a7958.

rqou, 2017-08-07:
Hmm, somehow this doesn't seem to be working for me. I have [this json](https://gist.github.com/rqou/28e84a8365771e2185f859b4f47ad206) (caution: huge file) and running opt_rmdff doesn't seem to do anything to the flip flop cells even though all of them should have a constant "0" on the set and reset inputs.

cliffordwolf, 2017-08-08:
Todo: add code to lower `$dffsr` and `$_DFFSR_*_` cells to non-sr types if they have only one (or zero) possible reset values.

cliffordwolf, 2017-08-09:
Implemented in 007f29b.

closed 2017-08-09 by cliffordwolf