YosysHQ/yosys #378: Feature request: Ability to infer toggle flip-flops

rqou, 2017-08-08:
Coolrunner-II has these as built-in primitives

cliffordwolf, 2017-08-09:
Robert, if you don't want me to close this right away you have to at least provide a sufficient description of the feature you want. Give me at least an example of the input you'd want to be able to process and the kind of output you'd expect.

If this implies the addition of a new cell type: what properties would this new cell type have?

rqou, 2017-08-09:
Apologies for not adding enough detail earlier. Yes, I would like to add a bunch of new cell types. As an example, I have the following Verilog model of the `FTCP` primitive in the Coolrunner-II CPLDs:
```verilog
module FTCP (C, PRE, CLR, T, Q);
    parameter INIT = 0;

    input C, PRE, CLR, T;
    output wire Q;
    reg Q_;

    initial begin
        Q_ <= INIT;
    end

    always @(posedge C, posedge PRE, posedge CLR) begin
        if (CLR == 1)
            Q_ <= 0;
        else if (PRE == 1)
            Q_ <= 1;
        else if (T == 1)
            Q_ <= ~Q_;
    end

    assign Q = Q_;
endmodule
```

This is a toggle flip-flop with set and reset. If the T input is high, the output of the flip-flop will toggle when there is a rising edge. If the T input is low, the output will not change. Right now, when yosys sees this construct, it creates a `$dffsr` cell with the output connected to a `$not` and a `$mux`, which feeds back into the input of the `$dffsr` cell. I would like to add some `$tff` and `$tffsr` cells (and the corresponding `$_TFF_*` and `$_TFFSR_*` cells) to represent this construct instead.

rqou, 2017-08-09:
I figured that adding toggle flip-flops as a yosys builtin cell type (rather than adding it as a Coolrunner-II-specific extraction pass) can potentially help any other devices that also have toggle flip-flops. It is always possible to lower them back into normal D flip-flops if the target does not have them as primitives.

rqou, 2017-08-11:
As an initial strawman proposal for how this might be implemented, I have written [this code](https://github.com/azonenberg/yosys/commit/fd83c69a21c27f47577ea904b1520dfbf0ad04fb) (and [bugfix](https://github.com/azonenberg/yosys/commit/94da51f707818d95cc75dfe2ff3ab803dce44257)). This code was written for the circuit reverse engineering project that @azonenberg and I are working on, but the code seems to correctly find toggle flip-flops for synthesis in the scenarios that I tested. However, the particular method of finding them (running ABC twice) seems like it might be suboptimal for synthesis.