YosysHQ/yosys #390: Question

OrDavidi, 2017-08-23:
I am trying to get a flat gate-level verilog from a top verilog design ( from multiple files.)
i have a folder / folders with verilog files and i would like the synthesis pass will run on the entire design.

i am running: 
read_verilog /home/ord/synth_tests/arb.v
hierarchy -check -libdir /home/ord/synth_tests/; proc; opt; techmap; opt
write_verilog synth.v

But i still got instances inside the output file ( synth.v ) which were not flattened and synthed.
what am i missing ? 

module arb.v(...)
...
mod_a inst_a(..); // **i expect this will be synth and written in the same file.**

endmodule

Thanks!

cliffordwolf, 2017-08-23:
Please post an [mcve](https://stackoverflow.com/help/mcve).

OrDavidi, 2017-08-23:
i can't reproduce on a small example - can i send you an email with my source files ?

OrDavidi, 2017-08-23:
Sorry my bad - i found the issue.

closed 2017-08-23 by OrDavidi