YosysHQ/yosys #391: Unexpected SMT solver error when dumping vcd file

banahogg, 2017-08-25:
Yosys 0.7+285 (git sha1 d3b3dd8, gcc 4.8.4-2ubuntu1~14.04.3 -fPIC -Os)

[bugrepro.zip](https://github.com/cliffordwolf/yosys/files/1250765/bugrepro.zip)

Occurs with either yices2 or z3

Result from running make in the attachment:

##      0   0:00:00  Assert failed in test_simp_decode_formal: test_simp_decode_formal.v:45
##      0   0:00:00  Writing trace to VCD file: dump.vcd
SMT Solver Error: (error "line 203 column 157: unknown function/constant test_simp_decode_formal_m:W0A test_counts")

It's possible this is caused by the testbench code itself, in which case this bug is a request for a clearer error message :-)

cliffordwolf, 2017-08-25:
Thanks for reporting this issue. Fixed in c2d7374.

closed 2017-08-25 by cliffordwolf

banahogg, 2017-08-26:
Confirmed fixed on my full test case - thanks.