YosysHQ/yosys #393: Feature request: optimize pass for gt cells with constant output

Kmanfi, 2017-08-28:
Here is test module for gt.
Same kind of optimisation is also possible for lt cells.

```
module gt_test(input [7:0] a, input [7:0] b, output y, output x, output z, output yn, output xn, output zn, output s, output gt);
  assign y = ({a[0], a[0], 2'b10} > {2'b00, a[0], a[0]}); // const 1 
  assign x = ({a[0], a[0], 2'b00} > {2'b10, a[0], a[0]}); // a[0]
  assign z = ({2'b10, b[0], a[0]} > {a[0], a[0], 2'b00}); // not(a[0])

  assign y = ({2'b00, a[0], a[0]} > {a[0], a[0], 2'b00}); // const 0
  assign x = ({2'b10, a[0], a[0]} > {a[0], a[0], 2'b00}); // not(a[0])
  assign z = ({a[0], a[0], 2'b00} > {2'b10, b[0], a[0]}); // not(a[0])

  assign yn = ({a[0], a[0], 2'b00} > {2'b00, a[0], a[0]}); // a[0]
  assign xn = ({a[0], a[0], 2'b00} > {2'b10, a[0], a[0]}); // a[0]
  assign zn = ({2'b10, b[0], a[0]} > {a[0], a[0], 2'b00}); // b[0] & not(a[0])

  assign s = ({a[0], a[0]} > 2'b11); // const 0

  assign gt = ({1'b1, 1'b1, a[0], 2'b11} > {1'b1, a[0], 1'b1, 1'b0, a[0]}); // const 1
endmodule
```

Kmanfi, 2019-05-18:
Above module for peep hole opt enthusiasts. 😃

cliffordwolf, 2019-05-18:
Note to anyone who wants to take a look at this: All single-cell optimizations (such as this one) should go into `opt_expr`, _not_ in a new pass.