YosysHQ/yosys #400: Feature request: recover_reduce to support single bit coarse gates $and, ...

Kmanfi, 2017-08-29:
Hello,
coarse gate designs have this same need. Currently I have 60.000 single bit $and and 60.000 single bit $or cells in some design. Majority of those form long reduce chains. It is important to reduce those chains because it might reduce cell count dramatically in some cases. Like in mine. There are some simulation situations where reduced cell count will affect to simulation speed. Chains might be very long so option to limit width of reduce cell will help to get a bit more realistic cell count in coarse level.

Kmanfi, 2017-08-29:
Here is test unit for this feature request. Now it will make three single bit $and. One 4bit wide $reduce_and would be nice.
```
module and_reduce (a, b, x);
input [3:0] a, b;
output [3:0] x;
    assign x = (a[3] & a[2] & a[1] & a[0]);
endmodule

```

povik, 2023-09-29:
There's the `extract_reduce` pass for transforming `$_AND_`, `$_OR_`, `$_XOR_` chains into reduce cells as you describe, but at the moment it doesn't consider single-bit `$and` cells and similar.