YosysHQ/yosys #414: Preprocessor expansion order (and handling of a `" token)

combinatorylogic, 2017-09-20:
This does not seem to work at the moment:

```verilog
`define STR(a) `"a`"
`define PATHPREFIX .
`define IPATH(a) `include `STR(`CPUPREFIX/a)
`IPATH(tst.v)
```

Apparently, handling of the `include directive currently only allows expansion of a macro without arguments.

combinatorylogic, 2017-09-21:
P.S. please see pull request #415 as an illustration of this issue.

cliffordwolf, 2017-09-26:
Thanks for reporting the issue *and* fixing it. :)

Afaict Icarus Verilog also has problems with the code you posted. Do you want to report that or should I?

closed 2017-09-26 by cliffordwolf

combinatorylogic, 2017-09-29:
Sorry, did not see this comment. Did not even try testing it with Icarus, only checked Vivado and ISE. Please feel free to report the issue!