YosysHQ/yosys #416: BLIF frontend limited to gates with 8 or less nets

heiner-bauer, 2017-09-25:
It looks like the BLIF frontend is deliberately limited to gates with 8 or less nets
https://github.com/cliffordwolf/yosys/blob/143c0abd33ed76b2a7e38dbbac1767e6f7edd68f/frontends/blif/blifparse.cc#L525 but I can not immediately see where this restriction comes from.

I stumbled over this issue while using an ABC pass, which internally uses `parse_blif()` to reintegrate ABC results. From the log file it was not immediately clear whether this was an issue with the ABC script
```
24.2.1. Executing ABC.
Running ABC command: <yosys-exe-dir>/yosys-abc -s -f <abc-temp-dir>/abc.script 2>&1
ABC: ABC command line: "source <abc-temp-dir>/abc.script".
ABC: 
ABC: + read_blif <abc-temp-dir>/input.blif 
ABC: + read_library <abc-temp-dir>/stdcells.genlib 
...
ABC: abc - > if -K 9;+ write_blif <abc-temp-dir>/output.blif 
ERROR: Syntax error in line 866!
```
 so I would also suggest adding a prefix to the error messages in `blifparse.cc`.

cliffordwolf, 2017-09-26:
> but I can not immediately see where this restriction comes from.

The exponential size of the lookup table generated. How many bits do you need?

heiner-bauer, 2017-09-27:
In my case 10 bits would suffice. 
However, i wonder if it is really necessary to enumerate the complete looukp table, ABC seems to work flawlessly up to 30 bits (`read_blif`, `if`, and `write_blif`).

cliffordwolf, 2017-09-27:
> In my case 10 bits would suffice.

I've now increased the limit to 12 bits in 3039627.

> However, I wonder if it is really necessary to enumerate the complete looukp table,

This is necessary by definition, because `read_blif` maps the BLIF `.name` blocks to LUTs.

Call `read_blif -sop` instead if you want to map to sum of products (see `help read_blif`). In this mode there is no limit on the number of inputs to a gate.

closed 2017-09-27 by cliffordwolf