YosysHQ/yosys #417: Problem parsing blif file

lfigiel, 2017-09-25:
Is it a problem with yosys or arachne-pnr ?
[cseed/arachne-pnr#75](https://github.com/cseed/arachne-pnr/issues/75)

cliffordwolf, 2017-09-26:
What's your input? What's your script? It can't reproduce this without this information.

See also Q1 in the [Yosys FAQ](http://www.clifford.at/yosys/faq.html).

cliffordwolf, 2017-09-26:
I'm closing this a a possible dup of #404.

closed 2017-09-26 by cliffordwolf

lfigiel, 2017-09-26:
Attached, please find the source code
[top.zip](https://github.com/cliffordwolf/yosys/files/1333115/top.zip)
To reproduce the issue:
yosys -q -p "synth_ice40 -blif top.blif" top.v
arachne-pnr -d 1k -P vq100 -p iceblink40.pcf top.blif -o top.txt

Please note that the code works perfectly in iCEcube2 for the same platform (iCE40).

lfigiel, 2017-09-26:
So my understanding is that yosys's output uses something that is not present in the hardware - hence the error unknown model `$DFFSR_PPP' but somehow iCEcube2 finds the way to synthesis same code for the same platform.

cliffordwolf, 2017-09-26:
Did you read the explanation in #404 that I linked to? I explained how they do it there.

lfigiel, 2017-09-26:
Yes, I did, it is all clear now, thank you. I will try to re-write the code to avoid using missing components.