YosysHQ/yosys #418: Building on Centos7

salmansheikh, 2017-09-25:
I got clang, python3, and I first configured for gcc but then did clang after installing it and after make, I get this error:. Any ideas why this is happening? 

[Makefile.conf] CONFIG := clang
[  0%] Building frontends/verilog/verilog_parser.tab.cc
frontends/verilog/verilog_parser.y:139.21-27: syntax error, unexpected identifier
make: *** [frontends/verilog/verilog_parser.tab.cc] Error 1

salmansheikh, 2017-09-25:
I was missing some libraries flex, bison and one other.

closed 2017-09-25 by salmansheikh

ninhho, 2020-02-13:
Hi salmansheikh,

I got the similar issue with yours 
[  9%] Building frontends/ilang/ilang_parser.tab.cc
frontends/ilang/ilang_parser.y:48.20-44: syntax error, unexpected {...}
frontends/ilang/Makefile.inc:8: recipe for target 'frontends/ilang/ilang_parser.tab.cc' failed
make: *** [frontends/ilang/ilang_parser.tab.cc] Error 1

bison (GNU Bison) 2.4.1
flex 2.5.35
cmake version 3.4.3
Is there any libraries that I missed? 
Could you please help me how did you get out of this? Thanks.

Ninh

nakengelhardt, 2020-02-13:
@ninhho have you installed all the libraries listed in the [Setup section of the readme](https://github.com/YosysHQ/yosys#setup)?

ninhho, 2020-02-15:
@nakengelhardt As far as I know, I installed below (in **Bold**)
**clang bison flex** \
	libreadline-dev gawk tcl-dev **libffi-dev** **git** \
	graphviz xdot **pkg-config** **python3** libboost-system-dev \
	libboost-python-dev libboost-filesystem-dev zlib1g-dev

How can I know each of them installed or not? Thank you.

daveshah1, 2020-02-15:
You probably want a newer version of bison - 2.4.1 looks to be from 2008!

ninhho, 2020-02-15:
@daveshah1 Thanks for advice, I have updated the newer version of bison 2.7
However the errors still there.
[  9%] Building frontends/ilang/ilang_parser.tab.cc
frontends/ilang/ilang_parser.y:48.20-44: error: syntax error, unexpected {...}
frontends/ilang/Makefile.inc:8: recipe for target 'frontends/ilang/ilang_parser.tab.cc' failed
make: *** [frontends/ilang/ilang_parser.tab.cc] Error 1
[den-l-003.amperecomputing.com]hninh@amperecomputing.com /scratch/den_scratch/hninh/chipyard/tools/yosys/>                                                                                                              >>> chipyard/tools/yosys/> vi README.md 
[den-l-003.amperecomputing.com]hninh@amperecomputing.com /scratch/den_scratch/hninh/chipyard/tools/yosys/>                                                                                                              >>> chipyard/tools/yosys/> bison --version
bison (GNU Bison) 2.7
Written by Robert Corbett and Richard Stallman.

Copyright (C) 2012 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

daveshah1, 2020-02-15:
I think that Yosys needs Bison 3+ to build

ninhho, 2020-02-15:
I will give it a try, will feedback you soon. Thanks.

ninhho, 2020-02-15:
@daveshah1 It looks like not completed yet. Maybe missing libffi...
[ 89%] Building techlibs/sf2/sf2_iobs.o
[100%] Building yosys
/usr/bin/ld: cannot find -lffi
collect2: error: ld returned 1 exit status
Makefile:614: recipe for target 'yosys' failed
make: *** [yosys] Error 1

Do you have any suggestion?

ninhho, 2020-02-15:
Hi @daveshah1 
I have updated the libffi and bypassed above issue. However I got another below.
[ 95%] ABC: `` Compiling: /src/sat/glucose/AbcGlucoseCmd.cpp
In file included from ./src/misc/vec/vec.h:29:0,
                 from ./src/aig/gia/gia.h:34,
                 from src/sat/glucose/AbcGlucoseCmd.cpp:21:
./src/misc/util/abc_global.h:125:5: error: #error unsupported platform
    #error unsupported platform
     ^
Makefile:182: recipe for target 'src/sat/glucose/AbcGlucoseCmd.o' failed
make[1]: *** [src/sat/glucose/AbcGlucoseCmd.o] Error 1
Makefile:663: recipe for target 'abc/abc-71f2b40' failed
make: *** [abc/abc-71f2b40] Error 2

Could you please take a look? Thanks.