YosysHQ/yosys #419: Potential issue with memories

rafaeltp, 2017-09-26:
Hi,

I have a simple test case for a memory and I am finding a odd behavior (in my understanding at least) when calling memory_collect.
The input is:

```verilog
module rf( input      [2:0]  wr_addr
          ,input      [7:0]  wr_data
          ,input             wr_e

          ,input      [2:0]  rd_addr
          ,output reg [7:0]  rd_data
          ,input             clk);

  reg [7:0] rf [7:0];

  always @ (posedge clk) begin
    rd_data <= rf[rd_addr];
    if(wr_e)
      rf[wr_addr] <= wr_data;
  end
endmodule
```

I am running the following script: 

```
yosys -p "read_verilog -sv test.v; proc; memory_collect; opt; write_verilog foo.v"
```

The output foo.v contains (I removed src file info for clarity):

```verilog
/* Generated by Yosys 0.7+316 (git sha1 2cc09161, clang 5.0.0 -fPIC -Os) */

module rf(wr_addr, wr_data, wr_e, rd_addr, rd_data, clk, ce);
  wire [2:0] _00_;
  wire [7:0] _01_;
  wire [7:0] _02_;
  wire [7:0] _03_;
  reg [2:0] _04_;
  reg [7:0] _05_;
  reg [7:0] _06_;
  input ce;
  input clk;
  input [2:0] rd_addr;
  output [7:0] rd_data;
  reg [7:0] rd_data;
  input [2:0] wr_addr;
  input [7:0] wr_data;
  input wr_e;
  always @(posedge clk)
      _04_ <= _00_;
  always @(posedge clk)
      _05_ <= _01_;
  always @(posedge clk)
      _06_ <= { _02_[7], _02_[7], _02_[7], _02_[7], _02_[7], _02_[7], _02_[7], _02_[7] };
  always @(posedge clk)
      rd_data <= _03_;
  assign _02_[7] = wr_e ? 1'h1 : 1'h0;
  assign _01_ = wr_e ? wr_data : 8'hxx;
  assign _00_ = wr_e ? wr_addr : 3'hx;
  reg [7:0] rf [7:0];
  always @(negedge 1'hx) begin
    if (_06_[0]) rf[_04_][0:0] <= _05_[0];
    if (_06_[1]) rf[_04_][1:1] <= _05_[1];
    if (_06_[2]) rf[_04_][2:2] <= _05_[2];
    if (_06_[3]) rf[_04_][3:3] <= _05_[3];
    if (_06_[4]) rf[_04_][4:4] <= _05_[4];
    if (_06_[5]) rf[_04_][5:5] <= _05_[5];
    if (_06_[6]) rf[_04_][6:6] <= _05_[6];
    if (_06_[7]) rf[_04_][7:7] <= _05_[7];
  end
  assign _03_ = rf[rd_addr];
endmodule

```

The issue I see here is in the writing part of the code, where there is no clock in the sensitivity list (`always @(negedge 1'hx)`). I would expect `posedge clk` instead.

A similar behavior is observed when not calling the `memory_collect` pass (only relevant part kept):

```verilog
// ...
  \$memrd  #(
    .ABITS(32'd3),
    .CLK_ENABLE(32'd0),
    .CLK_POLARITY(32'd0),
    .MEMID("\\rf"),
    .TRANSPARENT(32'd0),
    .WIDTH(32'd8)
  ) _07_ (
    .ADDR(rd_addr),
    .CLK(1'hx),
    .DATA(_03_),
    .EN(1'hx)
  );
  (* src = "test.v:21" *)
  \$memwr  #(
    .ABITS(32'd3),
    .CLK_ENABLE(32'd0),
    .CLK_POLARITY(32'd0),
    .MEMID("\\rf"),
    .PRIORITY(32'd8),
    .WIDTH(32'd8)
  ) _08_ (
    .ADDR(_04_),
    .CLK(1'hx),
    .DATA(_05_),
    .EN(_06_)
  );
// ...
```

However, `memory` seems to produce a valid output (only relevant parts):

```verilog
// ...
  always @(posedge clk)
      rd_data <= _024_;
// ...
  always @(posedge clk)
      \rf[0]  <= { _096_, _095_, _094_, _093_, _092_, _091_, _090_, _089_ };
  always @(posedge clk)
      \rf[1]  <= { _104_, _103_, _102_, _101_, _100_, _099_, _098_, _097_ };
  always @(posedge clk)
      \rf[2]  <= { _112_, _111_, _110_, _109_, _108_, _107_, _106_, _105_ };
  always @(posedge clk)
      \rf[3]  <= { _120_, _119_, _118_, _117_, _116_, _115_, _114_, _113_ };
  always @(posedge clk)
      \rf[4]  <= { _128_, _127_, _126_, _125_, _124_, _123_, _122_, _121_ };
  always @(posedge clk)
      \rf[5]  <= { _136_, _135_, _134_, _133_, _132_, _131_, _130_, _129_ };
  always @(posedge clk)
      \rf[6]  <= { _144_, _143_, _142_, _141_, _140_, _139_, _138_, _137_ };
  always @(posedge clk)
      \rf[7]  <= { _152_, _151_, _150_, _149_, _148_, _147_, _146_, _145_ };
// ...
```

cliffordwolf, 2017-09-26:
Why are you running `memory_collect` without first running `memory_dff`? Why call `memory_collect` directly at all and not simply call the `memory` pass?

rafaeltp, 2017-09-26:
hmmm.. I assumed `memory_dff` would map memories to dff (which I now realized is done by `memory_map`). Both the combo `memory_dff; memory_collect` and `memory -nomap` yield the behavior I was going for.
My bad for assuming and not properly reading the doc. And thanks for the prompt reply.

closed 2017-09-26 by cliffordwolf