YosysHQ/yosys #420: Error when compiling: duplicate struct timespec

sjalloq, 2017-09-27:
Hi there,

I'm trying to compile the head of Git on mingw64 and am hitting the following error.
Hash: 30396270a28471dc8832a93732541a79b61042bd

Thanks, Shareef.

`
In file included from C:/msys64/usr/local/src/yosys/abc/src/base/cmd/cmdAuto.c:33:0:
C:/msys64/usr/local/src/yosys/abc/lib/pthread.h: At top level:
C:/msys64/usr/local/src/yosys/abc/lib/pthread.h:308:8: error: redefinition of 'struct timespec'
 struct timespec {
        ^~~~~~~~
In file included from C:/msys64/mingw64/x86_64-w64-mingw32/include/time.h:293:0,
                 from C:/msys64/usr/local/src/yosys/abc/src/misc/util/abc_global.h:72,
                 from C:/msys64/usr/local/src/yosys/abc/src/base/cmd/cmdAuto.c:25:
C:/msys64/mingw64/x86_64-w64-mingw32/include/sys/timeb.h:90:8: note: originally defined here
 struct timespec {
        ^~~~~~~~
make[3]: *** [CMakeFiles/abc.dir/build.make:2487: CMakeFiles/abc.dir/src/base/cmd/cmdAuto.c.obj] Error 1
make[2]: *** [CMakeFiles/Makefile2:68: CMakeFiles/abc.dir/all] Error 2
make[1]: *** [Makefile:84: all] Error 2
make: *** [Makefile:423: abc/abc-cd6984ee82d4.exe] Error 2
`

cliffordwolf, 2017-09-27:
@rqou will look into it: https://twitter.com/rqou_/status/913097758055137280

facchinm, 2017-09-27:
The msys2 build (cross-)compiles fine from Arch with `i686-w64-mingw32-gcc (GCC) 7.2.0`
Attaching the build, in case you want to test it :wink: 
[yosis.zip](https://github.com/cliffordwolf/yosys/files/1338648/yosis.zip)

sjalloq, 2017-09-27:
OK, thanks.  Can I just say you're talking to a hardware engineer so .... say what?  ;-). I can type 'make'.

I'll test it!

rqou, 2017-09-28:
@sjalloq Are you using a `Makefile.conf` or otherwise setting `CONFIG`? For mingw/msys/other Windows builds, you need to set `CONFIG` by running e.g. `make CONFIG=msys2` (`mxe` also works but triggers slightly different hardcoded paths). This activates special code paths in the yosys Makefile that ends up defining `HAVE_STRUCT_TIMESPEC` for abc, which seems like it should fix your issue.

sjalloq, 2017-09-28:
I followed these steps: http://grbd.github.io/posts/2016/09/12/setting-up-the-icestorm-fpga-tools-for-windows/

rqou, 2017-09-28:
@sjalloq If you're just trying to get started with IceStorm+Yosys, and you don't mind using binaries compiled by "some random person on the internet," you can use the pre-compiled binaries at https://rqou.com/jenkins/job/open-fpga-tools/ . You can grab the latest successful builds (which are from about a week ago; I'm working on it). As a disclaimer, I am not officially affiliated with the yosys project in any way, and these binaries are provided on a best-effort basis.

As for the instructions themselves, the instructions you linked recommend applying a patch to the yosys makefile (reproduced for posterity below). For some reason, that patch causes abc to be built once with make and once with cmake. I'm not aware of anybody who uses or tests the cmake build script for abc, so that might be the cause of the problem. Have you tried building without applying that patch?

```patch
--- yosys/Makefile	2016-09-12 19:52:11.801560700 +0100
+++ yosys.orig/Makefile	2016-09-12 19:47:23.501827700 +0100
@@ -175,8 +175,8 @@
 EXE = .exe
 
 else ifeq ($(CONFIG),msys2)
-CXX = i686-w64-mingw32-gcc
-LD = i686-w64-mingw32-gcc
+CXX = gcc
+LD = gcc
 CXXFLAGS += -std=c++11 -Os -D_POSIX_SOURCE -DYOSYS_WIN32_UNIX_DIR
 CXXFLAGS := $(filter-out -fPIC,$(CXXFLAGS))
 LDFLAGS := $(filter-out -rdynamic,$(LDFLAGS)) -s
@@ -391,11 +391,12 @@
 		test $(ABCPULL) -ne 0 || { echo 'REEBE: NOP abg hc gb qngr naq NOPCHYY frg gb 0 va Znxrsvyr!' | tr 'A-Za-z' 'N-ZA-Mn-za-m'; exit 1; }; \
 		echo "Pulling ABC from $(ABCURL):"; set -x; \
 		test -d abc || hg clone $(ABCURL) abc; \
-		cd abc && $(MAKE) DEP= clean && hg pull && hg update -r $(ABCREV); \
+		cd abc && $(MAKE) DEP= clean && hg pull && hg update -r $(ABCREV) && cmake ./ ; \
 	fi
 endif
 	$(Q) rm -f abc/abc-[0-9a-f]*
 	$(Q) cd abc && $(MAKE) $(S) $(ABCMKARGS) PROG="abc-$(ABCREV)$(EXE)" MSG_PREFIX="$(eval P_OFFSET = 5)$(call P_SHOW)$(eval P_OFFSET = 10) ABC: "
+	${Q} cp abc/abc$(EXE) abc/abc-$(ABCREV)$(EXE)
 
 ifeq ($(ABCREV),default)
 .PHONY: abc/abc-$(ABCREV)$(EXE)
```

mmicko, 2020-03-12:
Closing this since msys2 can successfully compile abc in 32bit and 64 configuration without problem

closed 2020-03-12 by mmicko