YosysHQ/yosys #422: No support for parameter arrays

sjalloq, 2017-09-28:
I've not been able to work out from the Verilog LRM if this is supported, but Verilator seems happy, although both Yosys and Icarus bomb out.

localparam integer PARAM_ARRAY [1:0] = '{1,0};

sjalloq, 2017-09-28:
I've worked around this using a function to select the correct param out of a concatenation.  Might be useful for anyone else trying to do this.

`function [31:0] paramselect;
  input integer sel;
  input [NUM_REGS*32-1:0] params;
  begin
    paramselect = (params >> sel*32);
  end
endfunction // paramselect
`

And I'm using that to set the reset value of a register file something like this:

`
parameter REG_RST_VAL_A = {0,1,2,3};

genvar g;
generate
    for(g=0; g<NUM_REGS; g=g+1)
      begin : g_regfile
          generic_reg
            #(.REG_RST_VAL (paramselect(g,REG_RST_VAL_A)))
              u_reg (.clk            (clk),
                          .reset_n        (reset_n),
                          .reg_wr         (cpu_wr_vec[g]),
                          .reg_wr_data    (cpu_wdata),
                          .reg_data       (reg_data[g]));
    end
  endgenerate
`

cliffordwolf, 2017-09-30:
I can't find any mention of this in the LRM. I'd assume it's SystemVerilog, not Verilog.

closed 2017-09-30 by cliffordwolf