YosysHQ/yosys #424: Some upstream abc portability issues

rqou, 2017-09-28:
https://bitbucket.org/alanmi/abc/issues/72/use-of-fpu_controlh-is-nonstandard-breaks

https://bitbucket.org/alanmi/abc/issues/73/commit-1f35664-is-wrong-breaks-build

Waiting for abc to pull these, and then yosys should probably update its pinned revision.

cliffordwolf, 2017-09-28:
Hahaha: https://github.com/cliffordwolf/yosys/blob/master/libs/minisat/00_PATCH_no_fpu_control.patch

All SAT solvers are MiniSAT. :) 

Just a few weeks ago I asked Alan to merge this fix into ABC's copy of glucose because that broke the build on Linux:

https://github.com/cliffordwolf/yosys/blob/master/libs/minisat/00_PATCH_mkLit_default_arg.patch

rqou, 2017-10-06:
@cliffordwolf can you pull this? https://bitbucket.org/alanmi/abc/commits/c454b12b7100b952fd2cee544dcec4d2299afc95

cliffordwolf, 2017-10-06:
Done in commit 2b04e8c. Thanks for taking care of this issue.

closed 2017-10-06 by cliffordwolf