YosysHQ/yosys #427: Use of SB_GB_IO

mubes, 2017-09-29:
First of all, I apologise for this 'cos I suspect strongly the problem is somewhere between my keyboard and my chair, however, I am having huge difficulties instantiating a SB_GB_IO.  The original clock comes from a pin on the device and I want to distribute it internally, so I thought this would be sufficient;

```
SB_GB_IO BtraceClk0
(
  .PACKAGE_PIN(traceClk),
  .GLOBAL_BUFFER_OUTPUT(BtraceClk)
);
```

however, when I try to synthesise this, I get;

```
read_blif tracIF.blif...
prune...
arachne-pnr: src/netlist.cc:657: void Model::check(const Design*) const: Assertion `!p2->is_bidir()' failed.
./create: line 2:  4153 Aborted                 (core dumped) arachne-pnr -d 8k -P ct256 -p toplevel.pcf tracIF.blif -o traceIF.txt
```

I'm afraid I'm a bit stuck. Sorry, I know this isn't a tech support forum, but the assertion barf means I don't really know how to move forward with this.

DAVE

cliffordwolf, 2017-09-29:
When an assert fires then usually there is a bug in the software, even when the correct behavior would be an error message that tells you what you've done wrong.

However, this seems to be an issue with arachne-pnr, not Yosys. Can you create an issue in the arachne-pnr issue tracker? It's here: https://github.com/cseed/arachne-pnr/issues/new

Also make sure to include a complete design that is actually synthesizable. From the assertion I would assume that the problem has something to do with one of the SB_GB_IO ports being connected to an IO pin that is declared `inout`, but there is no way of knowing because you have not posted sufficient information to actually reproduce the issue.

closed 2017-09-29 by cliffordwolf

mubes, 2017-09-29:
Thx. Your comments have given me a clue (I didn't realise GBINs were different to regular pins and had just allocated them randomly...I'm new here). Changing to a GBIN hasn't fixed the problem although it is a dedicated input only pin, not an inout.... but I will now investigate further before raising it as an issue at arachne-pnr with documentation for reproduction.