YosysHQ/yosys #444: Segmentation Fault with modified `demo2.v` from `examples/smtbmc`

cr1901, 2017-10-25:
I was modifying `demo2.v` from `yosys`' `examples/smtbmc` directory, when I discovered an input verilog file and commands which crash `yosys` with a Segmentation Fault. This crash is reproducible on both Windows and Linux.

First, the sample file:
```
// Nothing to prove in this demo.
// Just an example for memories, vcd dumps and vlog testbench dumps.

`ifdef FORMAL
`define assume(_expr_) assume(_expr_)
`else
`define assume(_expr_)
`endif

module demo2(input clk, input [4:0] addr, output reg [31:0] data);
	reg [31:0] mem [0:31];
	always @(posedge clk)
		data <= mem[addr];

	reg [31:0] used_addr = 0;
	reg [31:0] used_dbits = 0;
	reg initstate = 1;

    always @(posedge clk) begin
        initstate <= 1;

        if (!initstate) begin
            `assume(data != 0);
            `assume((used_dbits & data) == 0);
        end
    end
endmodule
```

This is a _contrived_ minimal example to reproduce the crash. I don't think it makes sense to simulate/verify/synthesize as-is.

Next the command invocation (`./yosys -p 'read_verilog -formal test.v; prep -top demo2'`):
```
 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2016  Clifford Wolf <clifford@clifford.at>           |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.7+361 (git sha1 1170508, gcc 5.4.0-6ubuntu1~16.04.4 -fPIC -Os)

-- Running command `read_verilog -formal test.v; prep -top demo2' --

1. Executing Verilog-2005 frontend.
Parsing formal Verilog input from `test.v' to AST representation.
Generating RTLIL representation for module `\demo2'.
Successfully finished Verilog frontend.

2. Executing PREP pass.

2.1. Executing HIERARCHY pass (managing design hierarchy).

[... 154 lines cut ...]
  Optimizing cells in module \demo2.
Performed a total of 0 changes.

2.6.12. Executing OPT_MERGE pass (detect identical cells).
Finding identical cells in module `\demo2'.
  Cell `$procdff$38' is identical to cell `$procdff$36'.
    Redirecting output \Q: $formal$test.v:24$2_CHECK = $formal$test.v:23$1_CHECK
    Removing $dff cell `$procdff$38' from module `\demo2'.
  Cell `$assume$test.v:24$15' is identical to cell `$assume$test.v:23$14'.
    Removing $assume cell `$assume$test.v:24$15' from module `\demo2'.
Removed a total of 2 cells.

2.6.13. Executing OPT_RMDFF pass (remove dff with constant values).
Removing $procdff$36 ($dff) from module demo2.
Removing $procdff$37 ($dff) from module demo2.
Segmentation Fault (core dumped)
```

Lastly when dumping the log to a file on Linux, I get this on the console (guessing an exception wasn't catched that was supposed to be?):

```
terminate called after throwing an instance of 'std::out_of_range'
  what():  dict::at()
Aborted (core dumped)
```

cliffordwolf, 2017-10-26:
Some kind of weird memory corruption.. I'm looking into it now.

Some notes for the record below. Feel free to ignore.

----

Minimal script to reproduce:
```
read_verilog -formal test.v; hierarchy; proc_init; proc_mux; proc_dff; proc_clean
opt_clean; opt_merge; opt_rmdff; opt_clean; opt_expr; opt_rmdff
```

Looks like a `Wire*` being deleted while still in use. Not sure if the bug even is in `opt_rmdff` or much earlier. The `std::out_of_range` exception is thrown by a method of a deleted container accessing freed data on the heap.

cliffordwolf, 2017-10-26:
Fixed in c238f45. Thanks for reporting this issue.

closed 2017-10-26 by cliffordwolf