YosysHQ/yosys #448: Spartan 3/6 support with ISE backend?

mithro, 2017-10-29:
What is required to be able to use Yosys with ISE for Spartan-6 and Spartan-3 devices?

I assume the following things are needed;
 1) [ ] Description of the black boxes like SERDES and such,
 2) [ ] Description of the technology for mapping
 3) [ ] Output format that Xilinx ISE can read?

Would I just follow the example in https://github.com/cliffordwolf/yosys/tree/master/techlibs/xilinx ? I think there where some recent pull requests for Altera stuff that I should be informative too?

If I understand correctly (3) might already be done?

I think there was also someone else interested in this?

dh73, 2017-10-30:
Hi @mithro ,

Much of the primitives from Xilinx' 7-series  shares the cell name an port numbers/functions that 6-series (both Spartan/Virtex). If you even miss some primitive such as IBUFGDS (differential buffer), you could instantiate it as a black box and both Yosys/XST will generate you a bitfile, you can take this as an example, both tested on HW:
**https://github.com/dh73/Sandbox/tree/master/blink_demo
https://github.com/dh73/Sandbox/tree/master/uart_demo**

About Spartan-3, for sure there will be some work to be done in the technology map flow. You can start by extracting the Spartan-3 simulation models from any ISE version and create a similar instantiation for Yosys techmap. Spartan-3 uses 4 input LUT, so you'll need to create a 4 input LUT instantiation module such as the ones you'll find in ice40 or intel cells_map.v

I added options to switch from different families in **https://github.com/dh73/yosys/blob/master/techlibs/intel/synth_intel.cc, you could use something similar.**

Let me know if you start working to accomplish this, and/or if you need any help.

gergoerdi, 2018-09-19:
Did anything ever come out of this? I'd like to start using Yosys to replace the front-end part of the Xilinx ISE pipeline, since it is [giving me grief](https://github.com/clash-lang/clash-compiler/issues/356).

mithro, 2018-09-21:
@gergoerdi Nobody has worked on this that I know of. Would love for someone to work on it :-)

Streetwalrus, 2019-04-25:
I've updated the old mojo example to work with yosys 0.8 and my S6 breakout board, you can find it [here](https://github.com/Streetwalrus/yosys-spartan6-example). It synthesizes and runs on hardware.
I'm not sure if there's anything in recent yosys versions that would break 6-series devices, but I suppose a few tweaks to the synthesis script should be all we need to fix it. In particular, does [the LUT setup](https://github.com/YosysHQ/yosys/blob/b2020ab44fe33b83dbde39fe1f0cd53e4460f2f0/techlibs/xilinx/synth_xilinx.cc#L307) need to be changed considering 6-series only have 6-input LUTs?

I know @cliffordwolf's stance is not to support 6-series devices because ISE is deprecated, but Xilinx themselves appear to be confused about Spartan 6's status: even though it's not supported in Vivado, I can't find any official statement from them saying that it's not recommended for new designs, and it's the only non-7-series family that's still being advertised all over their website, and they don't have any parts to fill the lowest cost roles.

I'm already looking into moving to ECP5 myself, since it's cheaper and larger than an LX9, but I would also love to contribute to the reversing effort to get support for Spartan 6 in nextpnr eventually, as I have a board and it would be a shame to let it rot away. :)

mwkmwkmwk, 2019-12-04:
Some updates:

- Spartan 6 should now be supported out of the box (`synth_xilinx -family xc6s -ise`), with the same support level as Series 7. If anything is not working as expected, it's a reportable bug.
- ISE output should also be working as expected (use `write_edif -pvector bra <file>.edif`), and should be usable for all families supported by both ISE and yosys (ie. Virtex 6, Spartan 6, Series 7, and some limitted WIP support for Virtex 5)
- Spartan 3* (and Virtex 2/4) support will happen at some point (it'd be very useful for me personally), and some initial work has been done (the blackbox cells are all there, and multiplier/DSP mapping is supported), but the whole mapping process needs to be changed to support LUT4 Xilinx devices — don't expect that to happen soon

mwkmwkmwk, 2019-12-04:
Tracking list for Spartan 3 support:

- #1547 
- #1548 
- #1549 
- #1550

Streetwalrus, 2019-12-04:
Great news! Thanks a lot for the work, this is very valuable.

rodrigomelo9, 2019-12-18:
Hi @mwkmwkmwk awesome work. I am new with Yosys. I know its existence for time ago, but this update inspired me to try :-D

I will report an issue where I want to help, to learn about how to improve the Xilinx support (to start with something :P). You help and further reading (part of the whole documentation that **I must** to read) will be welcome.

And maybe this issue is ready to be closed? Now is time to try and report issues :-D

mwkmwkmwk, 2020-02-07:
Update: initial support for LUT4 devices (including Spartan 3*) has just landed in master. This includes general logic, wide LUTs, carry chains, LUT RAM, block RAM.  Things still missing (in comparison to Spartan 6 / Series 7):

- [ ] shift register inference (#1548 )
- [ ] wide mux mapping (which needs to be redone in a more generic way anyhow)
- [ ] ABC9 support (needs timing models)

interfect, 2024-08-26:
I've discovered that the `-pvector bra` option to `write_edif` seems to be absolutely essential for creating an EDIF that Xilinx ISE can understand correctly. If I leave it out like [the example I was working from does](https://github.com/9ary/yosys-spartan6-example/blob/618328a594641ed27a6648118c20a5e6be1da99c/synthesize.sh#L30), ISE still *thinks* it has read the EDIF correctly, and I get a final bitstream that I can load and that sort of works.

But without `-pvector bra` it can't do math! Using "+" in my VHDL input (via the GHDL Yosys plugin) ends up getting the wrong answers (my design seems to think that 3 + 5 = 6, for example), but combinatorial logic still seems to work and I can still get switches on my dev board to control LEDs. Maybe something is going wrong with the ALU cells or something.

Is there any hope of getting some kind of warning message out of `write_edif` if the design it is going to write came from `synth_xilinx`, but the "wrong" output settings are going to be used?