YosysHQ/yosys #449: Failure when assert/assume'ing signals will only change on posedge of a clock

ZipCPU, 2017-10-30:
I'm getting the [same vs assert/assume problem](https://github.com/cliffordwolf/yosys/issues/447) with constructs such as:
```
always @($global_clock)
if (!$rose(i_clk))
begin
     // etc.
end

Not sure if it's the same issue as the last one or not, but this one requires a [new test case](http://zipcpu.com/tmp/ybug2.tgz).

Dan

cliffordwolf, 2017-10-30:
In the first cycle `$rose(i_clk)` is undefined. If you have multiple in your design then they are undefined independently of each other. Because of that, the assumption in your top level can be inactive in the first cycle and the assertion in the sub module can be active at the same time.

I'm not entirely sure what the best solution to this is. In most real-world examples you have a reset signal that is assumed to be high for the first cycle (or cycles), and all assertions are usually disabled for as long as the reset is active. This way we usually don't need to worry about the value of things like `$rose` and `$stable` in the first cycle.

ZipCPU, 2017-10-30:
Okay, got it.  Looks like I can then get around this problem via:

```
reg f_past_gbl_clock_valid;
initial f_past_gbl_clock_valid = 0;
always @($global_clock)
  f_past_gbl_clock_valid <= 1'b1;

always @($global_clock)
  if ((f_past_gbl_clock_valid)&&(!$rose(i_clk)))
  begin
     assert($stable(value));
  end
```

Let's close this for now.  If this doesn't work, we can re-open it later.

closed 2017-10-30 by ZipCPU