YosysHQ/yosys #450: VCD file doesn't follow logic in the code

ZipCPU, 2017-11-08:
Basically, upon a contradiction, it seems that yosys-smtbmc may or may not produce logic matching the code.  It may pick something that matches the assertions instead.

Example: http://zipcpu.com/tmp/ybug5.tgz

Dan

cliffordwolf, 2017-11-09:
Can you explain in what way the trace in the VCD violates the constraints in the design?

[image]

Afaics design constraints are as following (and they are all satisfied):

Initially a=0, b=1, c=0. That seems to be the case.

c is a delayed version of a. Both are constant zero, so that is also fine.

a, b,c must be stable on state transitions that do not have a positive clock edge. This is also the case: b only changes at positive clock edges.

ZipCPU, 2017-11-09:
Sigh.  I'm not seeing it either now.  I'll reopen this when/if I can find a better example.
Sorry, and thanks.
Dan

closed 2017-11-09 by ZipCPU