YosysHQ/yosys #456: Request for new cmd.

Kmanfi, 2017-11-21:
Hi, here is request for command that will combine single bits to busses(wide wires). Example code contains single bit wires t0, t1, t2, t3 that will ultimately be part of wider bus with tmp1 wires. This would make easier to merge single bit cells to wider cells. Like one 8 bit wide cell (e & f). Reducing total cell count would have speed improvement in simulation speed. I started to make the command myself, it works fine with all simple cases. But I got stuck cells with concatenations on inputs. So I ask help.
```
module and_reduce_get_bit (a, b, c, d, e, f, y, x, z, n);
input [7:0] a, b, c, d, e, f;
output [7:0] y, x, z;
output n;
wire [7:0] tmp1;
wire [3:0] tmp2;
wire [7:0] tmp3;

wire t0, t1, t2, t3;

    assign tmp1[7] = e[7] & f[7];
    assign tmp1[6] = e[6] & f[6];
    assign tmp1[5] = e[5] & f[5];
    assign tmp1[4] = e[4] & f[4];
//    assign tmp1[3] = e[3] & f[3];
//    assign tmp1[2] = e[2] & f[2];
//    assign tmp1[1] = e[1] & f[1];
//    assign tmp1[0] = e[0] & f[0];

    assign t3 = e[3] & f[3];
    assign t2 = e[2] & f[2];
    assign t1 = e[1] & f[1];
    assign t0 = e[0] & f[0];

    assign tmp2[3:2] = c[3:2] & b[3:2];
    assign tmp2[1:0] = {t1, t0} & d[1:0];

    assign y[7] = a[7] & tmp1[7];
    assign y[6] = a[6] & tmp1[6];
    assign y[5] = a[5] & tmp1[5];
    assign y[4] = a[4] & tmp1[4];
    assign y[3] = a[3] & tmp2[3];
    assign y[2] = a[2] & tmp2[2];
    assign y[1] = a[1] & tmp2[1];
    assign y[0] = a[0] & tmp2[0];

    assign x[7] = a[7] | tmp1[7];
    assign x[6] = a[6] | tmp1[6];
    assign x[5:3] = a[5:3] | {tmp1[5:4], t3};
    assign x[2:0] = a[2:0] | tmp2[2:0];

    assign tmp3 = b & {tmp1[7:4], {2'b11} ,t1, t0};
    assign z = d | tmp3;
    assign n = !t2;

endmodule
```

ZipCPU, 2017-11-23:
Why not just declare wire [3:0] t, and then access t[0], t[1], t[2], and t[3] instead of t0, t1, t2, t3?
Dan

Kmanfi, 2017-12-19:
For example. Above code might be machine generated.

lethalbit, 2022-08-16:
The utility of a command to do this particular task is minimal at best, bit grouping should  not meaningfully change the result of synthesis, and if the code is machine generated, then a wrapper module could easily be written to do the job of combining the buses if need be.

closed 2022-08-16 by lethalbit