YosysHQ/yosys #458: priority mux - 0 cells

drom, 2017-11-22:
```
Linux x64
Yosys 0.7+371 (git sha1 4782d59a, clang 4.0.1 -fPIC -Os)
```

Simple priority mux (https://github.com/drom/icicles/blob/master/src/prio.v):

```verilog
module prio (
    input sa, sb, sc,
    input [7:0] a, b, c, d,
    output reg [7:0] ou
);

always @*
    if (sa) ou = a;
    else if (sb) ou = b;
        else if (sc) ou = c;
                else   ou = d;

endmodule
```

Synthesis script (https://github.com/drom/icicles/blob/master/src/prio.tcl):
```tcl
read_verilog prio.v
hierarchy -check -top prio
write_json prio.rtl.json
```
Output JSON file has 0 cells:

```json
...
      },
      "cells": {
      },
      "netnames": {
...
```
Expected at least 3 `$mux` cells.

Kmanfi, 2017-11-22:
Please try with.
```
read_verilog prio.v
hierarchy -check -top prio
prep
opt
write_json prio.rtl.json
```

drom, 2017-11-22:
@Kmanfi that looks way better! What did I do wrong?

cliffordwolf, 2017-11-22:
@drom You need to run `proc`, directly or indirectly, to convert "processes" into netlist elements. It is one of the first commands that `prep` calls. Whatever else you do in your synthesis script, if you do not call `hierarchy` and `proc` then you should understand exactly what those commands would do and why you can leave them out in your case. Otherwise just call `hierarchy` and `proc`, usually as the first two command in your script after reading the HDL input files. (`chparam` is one of the very few commands that one might want to call before `hierarchy`, but most script don't need it.)

closed 2017-11-22 by cliffordwolf

cliffordwolf, 2017-11-22:
@drom If your design contains memories you also need to make sure that you call the `memory` pass, with `-nomap` if you do not want the pass to convert memories to logic and FFs.

So the short answer is: Use `prep`, `synth`, or `synth_*` to process your design. When you write your own script instead, make sure you understand what the commands in `prep` do before you make the decision to not include them in your script.

drom, 2017-11-22:
Thank you for the explanation. My goal was to read verilog design and dump the highest level intermediate representation before any lowering steps. I saw that logical expressions like `&` or `|` getting into cells, but `if` statements are not. That was my confusion.