YosysHQ/yosys #459: generate'd variables don't show up in VCD file

ZipCPU, 2017-11-23:
Consider the source code and VCD found in http://zipcpu.com/tmp/ybug6.tgz.  When viewing the VCD file with gtkwave, the variables created internal to the generate statements are not shown.

Dan

ZipCPU, 2017-11-23:
Looking deeper, it looks like the variables will show up if the generate loop is named.

Dan

cliffordwolf, 2017-11-23:
Blocks without names without get assigned automatic names that by default do not show up in the VCD output. Name the generate block like this:

```
        generate for(k=0; k<NSIG; k=k+1)
        begin:myblock
            ....
        end
```

closed 2017-11-23 by cliffordwolf

AlexDaniel, 2017-11-23:
Oh. I think I've stumbled upon this some time ago, but wasn't able to figure out a solution. Maybe that should be added to [FAQ](http://www.clifford.at/yosys/faq.html)?