YosysHQ/yosys #466: Consider adding a Dockerfile to make it easier to build required image

dkarchmer, 2017-12-06:
Docker would allow people to build the code without having to worry about installing all the required dependencies.

This will also make it easy to migrate to different build systems, many which support Docker these days.

1138-4EB, 2017-12-29:
This might be useful to you: https://github.com/1138-4EB/elide/tree/master/docker

cliffordwolf, 2018-01-07:
WIP: https://hub.docker.com/r/openfpga/icetools/

hutch31, 2018-02-23:
Used the attached Vagrant configuration to successfully build and install yosys, useful if your native platform is non-Linux.

[yosys_vagrant.zip](https://github.com/YosysHQ/yosys/files/1750137/yosys_vagrant.zip)

abdelrahmanhosny, 2018-09-06:
@cliffordwolf the WIP link is not working. Have you managed to get a release of it?

daveshah1, 2019-06-26:
A Dockerfile has been added in #829

closed 2019-06-26 by daveshah1