YosysHQ/yosys #472: dfflibmap does not honor dont_use attribute

hutch31, 2017-12-19:
When using a liberty file where certain cells have a "dont_use : true;" attribute set on them, yosys does not honor the attribute and uses the marked cells.

cliffordwolf, 2017-12-25:
What version of Yosys are you using? Support for `dont_use` has been added two and a half years ago in 522705cc (2015-05-31). Please post an example liberty file that demonstrates the issue if you are using a version newer than that.

hutch31, 2017-12-27:
The version I am using is 8e22e811.  I later looked through the source and saw that dont_use support is present; naturally the library I compiled against is proprietary so I cannot submit it.  I will try to create a synthetic test case that reproduces this.

hutch31, 2018-01-19:
Unable to reproduce, will re-open if I can create a test case.

closed 2018-01-19 by hutch31