YosysHQ/yosys #474: Yosys segfaults if you run `verilog_defaults` command without doing anything else

mithro, 2017-12-24:
```
$ yosys

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2016  Clifford Wolf <clifford@clifford.at>           |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.7+381 (git sha1 8f2638ae, gcc 6.3.0-18 -fPIC -Os)

yosys> verilog_defaults 
Segmentation fault
```

mithro, 2017-12-24:
```
Program received signal SIGSEGV, Segmentation fault.
0x00007ffff66c68f4 in ?? () from /lib/x86_64-linux-gnu/libc.so.6
(gdb) bt
#0  0x00007ffff66c68f4 in ?? () from /lib/x86_64-linux-gnu/libc.so.6
#1  0x00007ffff7b76c4a in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::compare(char const*) const () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#2  0x0000555555640c6b in std::operator==<char, std::char_traits<char>, std::allocator<char> > (__lhs=..., __rhs=__rhs@entry=0x5555559b6cb3 "-add") at /usr/include/c++/6/bits/basic_string.h:5101
#3  0x00005555556d0829 in Yosys::VerilogDefaults::execute (this=this@entry=0x555555ce2980 <Yosys::VerilogDefaults>, args=std::vector of length 1, capacity 1 = {...}) at frontends/verilog/verilog_frontend.cc:413
#4  0x000055555563f903 in Yosys::Pass::call (design=design@entry=0x555555d1c610, args=std::vector of length 1, capacity 1 = {...}) at kernel/register.cc:232
#5  0x000055555563fc85 in Yosys::Pass::call (design=design@entry=0x555555d1c610, command="verilog_defaults") at kernel/register.cc:212
#6  0x00005555556810fc in Yosys::shell (design=0x555555d1c610) at kernel/yosys.cc:1067
#7  0x000055555562869d in main (argc=1, argv=0x7fffffffde78) at kernel/driver.cc:421
(gdb) 
```

cliffordwolf, 2017-12-24:
Haha. Fixed in 3400534.

closed 2017-12-24 by cliffordwolf