YosysHQ/yosys #475: Comment capability in scripts

ZipCPU, 2017-12-29:
The current script processing code for yosys has no comment capability.  One can be added via,

```diff --git a/kernel/yosys.cc b/kernel/yosys.cc
index 34665a0..c075fee 100644
--- a/kernel/yosys.cc
+++ b/kernel/yosys.cc
@@ -855,7 +855,7 @@ void run_frontend(std::string filename, std::string command, std::string *backen
                                        Pass::call(design, command);
                        }
 
-                       if (!command.empty()) {
+                       if ((!command.empty())&&(command[0] != '#')) {
                                handle_label(command, from_to_active, run_from, run_to);
                                if (from_to_active)
                                        Pass::call(design, command);
```

Thoughts or comments?

Dan

cliffordwolf, 2017-12-31:
Comments (with `#`) should work. That is handled in `Pass:call()`. Also `:...` can be used to comment-out a single command (instead of everything until the end of the line).

Do you have an example where comments don't work as expected?

ZipCPU, 2018-01-02:
Ok, I checked it today, it does work.  I'll close this with this comment.  Thanks,
Dan

closed 2018-01-02 by ZipCPU