YosysHQ/yosys #482: “CMOS Gate-Level Netlist” example segfaults on abc step

AlexDaniel, 2018-01-07:
This was done by following the example on http://www.clifford.at/yosys/screenshots.html precisely.
```
…
yosys> abc -liberty cmos_cells.lib

13. Executing ABC pass (technology mapping using ABC).

13.1. Extracting gate netlist of module `\counter' to `<abc-temp-dir>/input.blif'..
Extracted 14 gates and 22 wires to a netlist network with 6 inputs and 4 outputs.

13.1.1. Executing ABC.
Running ABC command: <yosys-exe-dir>/yosys-abc -s -f <abc-temp-dir>/abc.script 2>&1
ABC: ABC command line: "source <abc-temp-dir>/abc.script".
ABC: 
ABC: + read_blif <abc-temp-dir>/input.blif 
ABC: + read_lib -w /home/alex/project/cmos_cells.lib 
ABC: Parsing finished successfully.  Parsing time =     0.00 sec
ABC: Warning: Templates are not defined.
ABC: Libery parser cannot read "time_unit".  Assuming   time_unit : "1ns".
ABC: Libery parser cannot read "capacitive_load_unit". Assuming   capacitive_load_unit(1, pf).
ABC: Scl_LibertyReadGenlib() skipped sequential cell "DFF".
ABC: Library "demo" from "/home/alex/project/cmos_cells.lib" has 3 cells (1 skipped: 1 seq; 0 tri-state; 0 no func).  Time =     0.00 sec
ABC: Memory =    0.00 MB. Time =     0.00 sec
ABC: Warnings: genlib library reader cannot detect the buffer gate.
ABC: Some parts of the supergate-based technology mapper may not work correctly.
ABC: + strash 
ABC: + ifraig 
ABC: + scorr 
ABC: Warning: The network is combinational (run "fraig" or "fraig_sweep").
ABC: + dc2 
ABC: + dretime 
ABC: + strash 
ABC: + &get -n 
ABC: + &dch -f 
ABC: + &nf 
ABC: Segmentation fault
ERROR: ABC: execution of command "/usr/local/bin/yosys-abc -s -f /tmp/yosys-abc-sXUWiw/abc.script 2>&1" failed: return code 139.
```

Not sure what's going on there, it could be an issue in abc itself. In any case, examples in the documentation should not segfault I think?

cliffordwolf, 2018-01-07:
Thanks for reporting this.

There are a few issues here. The first is that we made some changes to the way we use ABC and the new way requires the cell library to contain a buffer cell. I've now updated the webpage accordingly.

The 2nd issue is that ABC segfaults instead of printing an error message and exiting with a nonzero return code. This is an issue in ABC and I reported it to the ABC maintainer.

AlexDaniel, 2018-01-07:
Is there a ticket ID or something?

cliffordwolf, 2018-01-07:
I sent an email, so no ticket ID. But I'll update this issue when I hear back from Alan.

AlexDaniel, 2018-01-08:
Are you sure it's entirely fixed? Because following the steps for the second screenshot, I get this:

[image]

That's almost on yosys HEAD, by the way.

cliffordwolf, 2018-01-08:
You have to load the library before running `show` (for example `read_liberty -lib <library_file>.lib`). Otherwise the `show` command has no way of knowing which pins are inputs and which are outputs.

AlexDaniel, 2018-01-08:
Right! Perhaps that could be included on the [screenshots](http://www.clifford.at/yosys/screenshots.html) page? So that examples are self-contained.

Also, I love you, yosys and all the contributors. Thank you very much for it! I don't work with yosys daily, but whenever I do it is so satisfying!

povik, 2023-09-29:
Going over old tickets, it looks like the original issue (ABC segfaulting on the example) is long gone. I will go ahead and close the ticket.

closed 2023-09-29 by povik