YosysHQ/yosys #484: (feature request) read_edif

AlexDaniel, 2018-01-08:
The other day I was working with some .edif files and realized that yosys has no `read_edif` command. I'm not sure if that is even possible or if there are any limitations, but at least that would be super helpful for visualizing the design with `show`. Don't know if that'd make it possible to use yosys for .edif → .blif conversion, but that'd be very cool, right?

To clarify, I'm talking about using yosys with .edif files that were generated by other software.

mithro, 2018-01-08:
We would love support for that!

https://github.com/SymbiFlow/ideas/issues/6

peterzieba, 2023-06-29:
Is it possible this was addressed with https://github.com/YosysHQ/yosys/pull/3499 ? It sounds like Yosys can be compiled with support for Verific, but if my understanding is correct, verific is not free? Are the necessary parts of it free?

nakengelhardt, 2023-06-30:
No, that is unfortunately very not free.