YosysHQ/yosys #485: yosys gives seg fault on large logic expressions

OrDavidi, 2018-01-14:
command : 
yosys -p read_verilog "g_mux__descaxi4stream_desc_agent_num4096.v"

the file attached in the zip file.

 Yosys 0.7+285 (git sha1 d3b3dd8, gcc 4.7.2 -O3 -DNDEBUG -fPIC)

-- Parsing `/usr/veri4/user/ord/verilog_for_debug/verilog/g_mux__descaxi4stream_desc_agent_num4096.v' using frontend `verilog' --

1. Executing Verilog-2005 frontend.
Parsing Verilog input from `/usr/veri4/user/ord/verilog_for_debug/verilog/g_mux__descaxi4stream_desc_agent_num4096.v' to AST representation.
Generating RTLIL representation for module `\g_mux__descaxi4stream_desc_agent_num4096'.
Warning: Deep recursion in AST simplifier.
Does this design contain insanely long expressions?
Segmentation fault

[g_mux__descaxi4stream_desc_agent_num4096.zip](https://github.com/YosysHQ/yosys/files/1629350/g_mux__descaxi4stream_desc_agent_num4096.zip)

Thanks

OrDavidi, 2018-01-15:
The stack overflowed on runtime.
In this case the recursion in AstNode::genRTLIL is huge ( > 32K ) so the big stack size terminates the run.

solved by adding to main : 
#include <sys/resource.h>
 const rlim_t kStackSize = 100L * 1024L * 1024L;   // min stack size = 100 Mb
  struct rlimit rl;
  int result;
  result = getrlimit(RLIMIT_STACK, &rl);
  if (result == 0)
    {
      if (rl.rlim_cur < kStackSize)
        {
          rl.rlim_cur = kStackSize;
          result = setrlimit(RLIMIT_STACK, &rl);
          if (result != 0)
            {
              fprintf(stderr, "setrlimit returned result = %d\n", result);
            }
        }
    }

cliffordwolf, 2018-01-16:
Any reason why you closed this issue? The recursive AST traversal caused issues with deep ASTs before and to me increasing the stack size seems like a reasonable way to deal with the issue.

OrDavidi, 2018-01-16:
closed by mistake when comment.

cliffordwolf, 2018-03-27:
Added code to increase the stack size in 0acea35. I hope that fixes the issue for you.

closed 2018-03-27 by cliffordwolf