YosysHQ/yosys #492: memrd cell type for firrtl backend not supported

edcote, 2018-01-22:
@azidar 

I am doing `read_verilog` followed by `write_firrtl`.  I get this error message:

`ERROR: Cell type not supported: $memwr (FinishQueue_1.$memwr$\ram$Top.DefaultConfig.v:74388$15603)`

      always @(posedge clk) begin
        if(reset) begin
          full <= 1'h0;
        end else if(T2) begin
          full <= do_enq;
        end
        if (do_enq)
          ram[1'h0] <= T6; // <-- right here
      end

`read_verilog` followed by `write_verilog` _does_ work.

Forgive my preliminary-only debug.  It does not appear that `verilog_backend.cc` does anything special something specific for `$memrd`:

See:

https://github.com/YosysHQ/yosys/blob/master/backends/verilog/verilog_backend.cc#L900
_vs_
https://github.com/YosysHQ/yosys/blob/master/backends/firrtl/firrtl.cc#322

There is mention of FIXME for `$memrd` in `verilog_backend.cc`
https://github.com/YosysHQ/yosys/blob/master/backends/verilog/verilog_backend.cc#L1114

(Assumption)  Perhaps there was an internal API/IR change that was not regressed on the firrtl backend?

azidar, 2018-01-22:
Yeah, there are many unsupported use cases - we haven't had a real user for this yet, so it's still very much in the alpha stage. I can't really commit much time at the moment to addressing this, but I will certainly review any PR's if you have time, and I believe @cliffordwolf would also accept a reviewed PR.

cliffordwolf, 2018-01-22:
What script are you using? Obviously you did not run `memory_collect` as part of your script. (Usually done via `memory`, which in turn is called by `prep`, which is the recommended meta command for stull like conversion to FIRRTL, BTOR, SMT, and other word-level formats.)

edcote, 2018-01-22:
Thanks for the prompt reply.  I can't commit to (help) to fix the issue right now, maybe next month?

The code looks relatively straightforward, but there is the ramp time :(

Do either of you have a proposed solution that I could implement in the code? Or, other pointers to reduce my ramp?  Alternatively, if either can provide a fix, I could write a handful of unit tests for the project.

edcote, 2018-01-22:
Adding `prep` (original issue did not use prep).

    yosys -import
    read_verilog Top.v
    hierarchy -check -top MyTop
    prep
    write_firrtl Top.fir

Results in new error: `ERROR: Cell type not supported: $shiftx [..]`

Will update title of issue.

edcote, 2018-01-22:
yosys -import
    read_verilog Top.v
    hierarchy -check -top MyTop
    memory # <-- seems to work better, I don't want optimization
    write_firrtl Top.fir
    
Results in new error: `ERROR: Cell type not supported: $pos [..]`

I worked around that issue.  Tried again.  Received error where the cell type not supported was the Verilog module itself.

    4. Executing FIRRTL backend.
    ERROR: Cell type not supported: CSRFile_0 (CsrUnit.CSRFile)

edcote, 2018-01-22:
Here is my nonsolution.  Enough to get me going for today.

https://github.com/YosysHQ/yosys/pull/494

Here is a summary of my observations.

    Warning: Unclocked write port 0 on memory RocketCAM_1.cam_tags.
    Warning: Negedge write port 0 on memory RocketCAM_1.cam_tags.
    Warning: Complex write enable on port 1 on memory RocketCAM_1.cam_tags.
    Warning: Cell type not supported: CsrUnit (RocketTile.CsrUnit)
    Warning: Cell type not supported: $pos (TLB_1.$extend$Top.DefaultConfig.v:69198$14494)
    Warning: Cell type not supported: $shiftx (TLB_1.$shiftx$Top.DefaultConfig.v:68425$14004)

cliffordwolf, 2018-01-23:
```
hierarchy -check -top MyTop
memory # <-- seems to work better, I don't want optimization
write_firrtl Top.fir
```

You are not even calling `proc`. This can't work. Please make sure that for all the commands in `prep` that you do not call you sufficiently understand what it does so you know it's safe to leave it out.

Also: The explicit purpose of `prep` is for cases where you don't want optimization. Otherwise you'd be using `synth` (or at least the `coarse:` part of it).

edcote, 2018-02-15:
Correct @cliffordwolf.  Closing this issue.  Now have better understanding of backend.  I will resolve the $mem issues with firrtl backend once I hit the problem again.

closed 2018-02-15 by edcote