YosysHQ/yosys #498: defparam failing inside generate

daveshah1, 2018-01-30:
While trying to synthesise some Lattice example code, I notice that Yosys will fail to process a defparam if the defparam and module it refers to are in a generate block.

A minimal example (iCE40) is:
```
module top(input a, output b);

generate
if (1) begin

SB_LUT4 lut_i (.O(b), .I0(a), .I1(1'b0), .I2(1'b0), .I3(1'b0));

defparam lut_i.LUT_INIT = 16'h2;

end
endgenerate

endmodule
```
Which fails with the following error:
```
1. Executing Verilog-2005 frontend.
Parsing Verilog input from `defparam.v' to AST representation.
Generating RTLIL representation for module `\top'.
ERROR: Can't find object for defparam `lut_i.LUT_INIT` at defparam.v:8!
```

zachjs, 2021-03-01:
I just re-ran this example, and it seems to be working correctly. Can you confirm? I'm guessing this was fixed as a side-effect of some of the other generate/scoping work.

@expipiplus1 It may also be worthwhile to run your generated code through and make sure that is working too.

gatecat, 2021-03-01:
Indeed the example case is working now, it's been a long 3 years and I don't think I'll find the original Lattice Verilog but I'll assume that is working fine now :)

expipiplus1, 2021-03-01:
Same for me!

closed 2021-03-03 by zachjs