YosysHQ/yosys #502: Select issue.

Kmanfi, 2018-02-05:
Hi,
can't select wire named \aa/N1
How to select such a wire?

```
read_ilang select.il 
dump
select -set testsel n:\aa/N1
dump @testsel
select -set testsel n:\data
dump @testsel
```

```
autoidx 1
attribute \top 1
module \wire_select
  wire \aa/N1
  wire input 1 \clk
  wire input 2 \data
  wire output 3 \out_o
  cell $dff $techmap\cc
    parameter \CLK_POLARITY 1'1
    parameter \WIDTH 1
    connect \CLK \aa/N1
    connect \D \data
    connect \Q \out_o
  end
  cell $not $techmap\cc1
    parameter \A_SIGNED 0
    parameter \A_WIDTH 1
    parameter \Y_WIDTH 1
    connect \A \clk
    connect \Y \aa/N1
  end
end
```

Kmanfi, 2018-02-13:
```
module top (A, B, C, Y);
    input    A, B, C;
    output   Y;

   MY_AND3 inst_t1 (.A(A), .B(B), .C(C), .Y(Y) );  

endmodule

module MY_AND3 (A, B, C, Y);
    input    A, B, C;
    output   Y;

//   MY_AND2 inst_a1 (.A(A), .B(B), .Y(\\SUM/N10) ); // will not read
   MY_AND2 inst_a1 (.A(A), .B(B), .Y( \\SUM/N10 ) ); // needs whitespaces around net name
   MY_AND2 inst_a2 (.A(C), .B( \\SUM/N10 ), .Y(Y) );

endmodule

module MY_AND2 (A, B, Y);
    input     A, B;
    output    Y;
  
assign Y = A & B;

endmodule

```

Kmanfi, 2018-02-13:
This is my current work around.
```
rename -hide
rename -enumerate w:*

```

cliffordwolf, 2018-02-13:
Fixed in 717abc9.

As work-around in older versions you can use this syntax: `*/n:aa/N1`

closed 2018-02-13 by cliffordwolf