YosysHQ/yosys #503: liberty support for bus, what extent supported? or bug?

edcote, 2018-02-07:
Earlier I hit this: https://stackoverflow.com/questions/47517707/yosys-read-liberty-syntax-error-on-pindatabus

Today:
    
    1. Executing Liberty frontend.
    ERROR: Missing or invalid direction for bus A on cell SP64x64_m4.

For a liberty file resembling this:

        bus (A) {
          bus_type : A_5_0;
          # direction label does not appear here
          pin ("A[5:0]") {
            related_power_pin : VDD;
            related_ground_pin : VSS;
            direction : input;

@cliffordwolf Are you looking at a specific spec/BNF for liberty cell format?  I would like to determine who is "at fault".  Then, I can decide on the next step.

edcote, 2018-02-07:
https://github.com/YosysHQ/yosys/blob/master/frontends/liberty/liberty.cc#L583-L586

Here is my initial hack.

    LibertyAst *dir = node->find("direction");
    if (!dir) dir = node->find("pin")->find("direction");
    if (!dir || (dir->value != "input" && dir->value != "output" && dir->value != "inout" && dir->value != "internal"))
        log_error([..])

edcote, 2018-02-15:
I am surviving with the following patch, a single line fix.

~~~~
diff --git a/frontends/liberty/liberty.cc b/frontends/liberty/liberty.cc
index 5458c60..8e9f8cc 100644
--- a/frontends/liberty/liberty.cc
+++ b/frontends/liberty/liberty.cc
@@ -581,6 +581,7 @@ struct LibertyFrontend : public Frontend {
                                                log_error("Error in cell %s: bus interfaces are only supported in -lib mode.\n",
 
                                        LibertyAst *dir = node->find("direction");
+                                       if (!dir) dir = node->find("pin")->find("direction");
 
                                        if (!dir || (dir->value != "input" && dir->value != "output" && dir->value != "inout" &&
                                                log_error("Missing or invalid direction for bus %s on cell %s.\n", node->args.at
~~~~

Recommend we keep this open someone else faces the issue or, perhaps, I hit another roadblock.

cliffordwolf, 2018-02-15:
Does 2c95dfc fix the issue for you?

edcote, 2018-02-15:
Fix is confirmed.  Closing.

closed 2018-02-15 by edcote